openscad/openscad #2501: "Hide editor" and "Hide console" reverts to true when closing OpenSCAD

korpx, 2018-10-05:
"Hide editor" and "Hide console" reverts to true in recent snapshots for Windows.

Unchecking these options in the GUI works, and the registry keys are immediately changed;
[HKEY_CURRENT_USER\SOFTWARE\OpenSCAD\OpenSCAD\view]
"hideConsole"="false"
"hideEditor"="false"

But when OpenSCAD is closed the options/registry is changed back;
[HKEY_CURRENT_USER\SOFTWARE\OpenSCAD\OpenSCAD\view]
"hideConsole"="true"
"hideEditor"="true"

Observed in several of the recent snapshots and as late as;
2018.10.05.ci118 (git 23b9526b)
OpenSCAD-2018.10.05.ci118-x86-64.zip
Windows 10 Enterprise build 1803

nophead, 2018-10-05:
Yes I see this too with version 2018.09.05 (git 40af343) that I downloaded as a snapshot but when I build master myself with MSYS2 I get version version 2018.09.26 and it works.

t-paul, 2018-10-05:
Hmm, looks like there's a stray `QSettings` in https://github.com/openscad/openscad/blob/master/src/openscad.cc#L661

nophead, 2018-10-05:
According to Blame it has been like that for years though. And why does it work when compiled with MSYS2?

t-paul, 2018-10-05:
All other places are changed to use a global `QSettingsCached` object. Could be a race condition between the two instances when closing the application.

t-paul, 2018-10-05:
Should be fixed by #2502.

closed 2018-10-06 by t-paul