openscad/openscad #2506: Verify and document behavior of function overrides

t-paul, 2018-10-07:
Followup on comment https://github.com/openscad/openscad/issues/1860#issuecomment-271954508...

This is likely a more general topic than just `assert()` as this potentially affects all the features which had lexer tokens added in the past, specifically all the list comprehension features.

> In OpenSCAD, I can overwrite build in functions. Example:
> function sin() = true;
> echo (sin(1));
> 
> but I can not overwrite assertion:
> function assert() = true;
> echo (assert(1));
> 
> ERROR: Parser error in file "overwrite.scad", line 1: syntax error
> 
> Is this a design feature of OpenSCAD that has to be documented as in "all OpenSCAD functions except the assert() function can be overwritten. The assert function can not be over written." or is it a bug in OpenSCAD?

MichaelAtOz, 2018-10-08:
I would like to be able to override functions (and modules) BUT with the ability to call the built-in's some way. Ala LISP Macros.

So, for example, you could have your cube() which did stuff then called (syntax e.g.) @cube(with your parameters), thus allowing you to customise all the "cube()'s" in the rest of the code.

If that was to happen, all functions (and modules) should be override-able.

I haven't used assert much, but being able to customise it in that manner could be very useful.

jordanbrown0, 2025-09-02:
Note that `echo()`, `assert()`, and `let()` are not truly functions.  They are funny-looking operators.  Similarly `for()`, `each`, `let()`, and `if()` in list comprehensions.  I would not be optimistic about overriding any of those.

You can already override true functions, but there's no way to call the underlying functions once you do.

Similarly for modules, though for them there may be more opportunities to override the keyword-based modules.

closed 2026-11-10 by github-actions[bot]