openscad/openscad #2507: Repaired imported STL keeps failing at render time

LatinSuD, 2018-10-07:
I was trying to add some legs to this STL I downloaded.
https://www.thingiverse.com/download:590084
It is a relatively simple model, but it keeps giving me errors after I render or do a "Check Validity".

The error varies and is something like:
`ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329 
`

Things I've tried so far:

- Stable version (2015.03-2, Linux and Windows) and Dev version (2018.09.05 git 40af343 Windows)
- Repair the STL with different services and application:
    - admesh
    - Netfabb
    - Freecad Analysis and repair mesh tools.
    - Meshlab (did the edge flip procedure among others)
- Tried compiling "edgeflip" version of OpenScad in Linux.

It always throws an error as long as I have more than one object in the scene.

Any clue what is happening?
I am no expert in OpenScad internals.

MichaelAtOz, 2018-10-08:
Basically CGAL, the library doing render (F6), requires a manifold STL (and other technical requirements*).
So when another object is there an implicit union() is performed, which calls CGAL.
CGAL doesn't like the broken STL.
I tried my usual repairs and failed too.
Then I tried https://tools3d.azurewebsites.net/ which is usually good, but it now outputs .3mf format.
I don't have anything that speaks 3mf.
So I give up, sorry.

* see [this discussion](http://forum.openscad.org/Why-are-manifolds-important-td23449.html)

LatinSuD, 2018-10-08:
I got the problem.
There are shared edges at the tail of the pig.
Sorry, I'm still a little newbie in this.

[image]

My question now is... Is there an easy way to move the shared edges? Using Meshlab or something?

t-paul, 2018-10-08:
I don't know an automated way for doing that. For manual fixes, I'm using Blender (which should find those edges via "select non-manifold").

t-paul, 2019-11-14:
Closing as OpenSCAD (by means of CGAL) can only support 2-manifold geometry, so shared edges can't be handled.

closed 2019-11-14 by t-paul