openscad/openscad #2510: Failed model withe relatively simple code.

UrielGuy, 2018-10-07:
Hey. 
The following model fails with the following error: 

> ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_3/SNC_external_structure.h Line: 1102 

`
$fn = 20; 

module side_pattern(h, twist_rate) {
    union() {
    linear_extrude(h, twist = 360 * (h / twist_rate)) 
    for (i = [1 : 5]) rotate([0,0,i * 360/10]) { square([0.5, 120], center = true); }
    linear_extrude(h, twist = -360 * (h / twist_rate)) 
    for (i = [1 : 5]) rotate([0,0,i * 360/10]) { square([0.5, 120], center = true); }
    }
}

union() {
    side_pattern(150, 200);
    sphere(100);
}
`

UrielGuy, 2018-10-08:
This also fails when using the most recent (2018-10-07) snapshot

MichaelAtOz, 2018-10-08:
Reproduced. Similar to last entry in #1525.

nophead, 2018-10-08:
It works if you have enough slices.

[image]

I think with too little slices the triangulation causes it to self intersect due to it being so thin. The thinner it or the greater the angle the more slices are required.

You can see the problem with one slice and a small twist.

[image]

Because the back and front triangulate from opposite corners the diagonal intersects. Possible if the triangles font and back were aligned it would work.

nophead, 2018-10-08:
Basically when you triangulate a quad there are  two possibilities. Depending on the twist one makes the face convex and the other makes it concave. If linear_extrude always chose the convex one I think it would avoid these self intersections. It actually seems to make the opposite choice because it flips when you twist in the other direction.

nophead, 2018-10-08:
With a shape like this there are four permutations triangulating the long faces of each slice. If they are both convex the shape will bulge. If one is convex and the other concave then I think the slice folds across the diagonal and the facets will be parallel giving constant thickness.  Only if both are concave then the slice narrows and can self intersect when thin enough but that seems to be what OpenSCAD chooses.

Another way to solve the problem is to break the long edges by adding more vertices. That is a lot faster than having a ridiculous number of slices. It looks like inline vertices get removed, so I had to perturbate them a tiny amount to force more quads.

[image]