openscad/openscad #2511: Font Burbank Big Cd Bd not visible in font lists

chris669, 2018-10-08:
Downloaded this font Burbank Big Cd Bd, I can use it in LibreOffice but when I click on Font List isn't there. BurbankBigCondensed-Bold.otf

t-paul, 2018-10-08:
Please state at least which Operating System and where/how the font is installed.

chris669, 2018-10-08:
Windows 10, downloaded the file, right clicked on it, clicked install and after checked if it was in c:\windows\font do not know how I could help more. As I told you, the font is visible in other programs like Libre Office.

MichaelAtOz, 2018-10-08:
Where did you download the font? I'll give it a try.

t-paul, 2018-10-08:
Can you please check if deleting the font cache folder like in https://github.com/openscad/openscad/issues/1892#issuecomment-292999791 helps?

chris669, 2018-10-09:
thanks, need this font for the kid (Fortnite font) :) https://www.fontsmarket.com/font-download/burbank-big-condensed-bold Tried also from another site, same file 110k Will try comment #1892 when in front of the PC, thanks, Chris

MichaelAtOz, 2018-10-09:
Works for me on Windows 7/64. Both 2015.03-2 & a recent snapshot.
[image]

Did you get this when you first ran OpenSCAD after installing the font?
[image]

chris669, 2018-10-09:
I checked #1892, and deleted the folder, then relaunch OpenSCAD, had the message "Fontconfig needs to update its font cache. then checked again but still not get the font.

[image]

[image]

FOR INFO: running latest Windows 10 October release.

MichaelAtOz, 2018-10-09:
Well that's my Windows fonts technical limit.
Perhaps try another font to see if it is just Burbank, or a bigger problem.
I came across this interesting one recently http://sansforgetica.rmit/

MichaelAtOz, 2018-10-09:
and I presume you can access Burbank from other apps? [an 'is it plugged in' question I know]

chris669, 2018-10-10:
Hi Michael, yes, can access from other apps, like LibreOffice or WordPad
[image]

chris669, 2018-10-10:
What I've seen also is they are different type fonts, like .ttc, ttf, this one is .otf. Restriction? And a strange thing is this is the one which is displayed  with an (french) accent "é" ("Burbank Big Gras condensé"). Maybe OpenSCAD doesn't manage special characters like other apps do?
[image]

MichaelAtOz, 2018-10-10:
@t-paul UNICODE?

MichaelAtOz, 2018-10-13:
@chris669 - Perhaps try another font to see if it is just Burbank, or a bigger problem.

Anyone else with Widows 10 & a language which uses character accents, perhaps you could try the font to see if it is Unicode or related?

t-paul, 2018-10-13:
I'm not aware of any Unicode issues that could prevent the font to be found at all. There's only #968 about display in the font list dialog.
As it's working on other systems it's probably at least connected to the local settings. Are there maybe some special characters in the path / name of the font file?

chris669, 2018-10-15:
Sorry away from the laptop for 1 week, will try another font asap. You can give me the link for different type of fonts that you want me to try.

MichaelAtOz, 2018-10-21:
> I came across this interesting one recently http://sansforgetica.rmit/

> What I've seen also is they are different type fonts, like .ttc, ttf, this one is .otf. Restriction? 

.tt* [truetype](https://en.wikipedia.org/wiki/TrueType) fonts, .otf [Opentype](https://en.wikipedia.org/wiki/OpenType) font. Both work with OpenSCAD. The Sansforgetica is a otf like burbank so probably a good test. You can remove it later.

> And a strange thing is this is the one which is displayed with an (french) accent "é" ("Burbank Big Gras condensé").

I found an Opentype font with accents in it's name, it worked in OpenSCAD (with a minor issue I'll report separately), it shows up in the OpenSCAD font list.

chris669, 2018-10-29:
Tried the Sans Forgetica, installed, deleted tempfont folder, and still the same issue: Not available  in OpenSCAD, but visible in LibreOffice.

[image]

MichaelAtOz, 2018-10-29:
@chris669 sorry I'm stumped. Possibly a Windows 10 issue.

t-paul, 2018-11-10:
Works without problem with Windows 10

[image]

chris669, 2018-11-10:
OK, so where's the problem? How did you installed the font?

t-paul, 2018-11-10:
I downloaded the font, right clicked it and selected "install". Next start of OpenSCAD had it available for use.

chris669, 2018-11-10:
ok, are you running the stable Openscad version or the latest build 2018? I'm running a version from 2 months ago, will check tomorrow...

t-paul, 2018-11-10:
I used build number [385](https://circleci.com/gh/openscad/openscad/385#artifacts).

chris669, 2018-11-11:
For me I have this: version 2018.09.05 (git 40af343)

chris669, 2018-11-12:
just installed the latest 2018.10.13 dev build 64 bits, still same problem. running Windows 10 Family 1809

rcmaniac25, 2019-07-29:
I know this is nearly a year old, but I'm having the same issue. I installed "Pathway Gothic One" on my Windows 10 (1903 build), am running 2019.07.28.ci3069 (git c31c9bfb) and have tried deleting the fontconfig cache multiple times, the font never shows up in the font list.

Word/Office, Wordpad, Notepad, Paint.net, Sublime Text, etc. have all found the font without any reload of the applications. I can't figure out where things are going wrong. I can't find any logs from fontconfig or FreeType within OpenSCAD would fail to load it. I've also spent maybe an hour going through the code and don't see anything in it, or the config files, that would filter out the font (though I'm not well versed in fontconfig configs).

Any ideas on what I can try to make this work?

Thanks for your time

MichaelAtOz, 2019-07-29:
@rcmaniac25 where did you download it from? (I'm still Windows 7 so can't see if it's a W10 problem)

rcmaniac25, 2019-07-29:
The font or the OpenSCAD build?

rcmaniac25, 2019-07-29:
Font: https://www.fontspace.com/eduardo-tunni/pathway-gothic-one
SCAD: http://www.openscad.org/downloads.html

MichaelAtOz, 2019-07-29:
Oh well, it works for me on W7, so like above it seems to point to W10.

We need a W10 tester...anyone?

rcmaniac25, 2019-07-29:
I'm not against testing stuff, so long as it doesn't take a long time. I don't really have the time to build things though at this moment.

MichaelAtOz, 2019-07-29:
Sorry, I meant a third party first to see if it is reproducible.

rcmaniac25, 2019-07-29:
Oh, ok.

MichaelAtOz, 2019-07-29:
> I came across this interesting one recently http://sansforgetica.rmit/

@rcmaniac25 The OP could not see sansforgetica in OpenSCAD, could you give it a try.

In my Windows Font list both it and your font have these same blank fields:

[image]

Also note the 'blocked' at the bottom of Right-click/Properties, I'm wondering if Windows 10 is being more protective? Try 'Unblock' if it's there on W10?

I'm wondering if it is badly formatted, as in the above list it shows as Font Name=...Light (note filename ...REGULAR) & Font Style=Light, but in OpenSCAD, Regular:

[image]

MichaelAtOz, 2019-07-29:
@rcmaniac25 Could you also try this.

At a command line enter
`openscad --debug FontCache,FontListDialog,FontListTableView,FreetypeRenderer` 
Then click `new` and paste this
`text("hello world",font="Pathway Gothic One"); // note - NO 'Regular'` and F5

Then post the console output.

chris669, 2019-07-29:
I tested also with this font, for me using latest Windows 10, I still cannot my Burbank font, even this one. Regards, Chris

MichaelAtOz, 2019-07-29:
@chris669 could you try the above --debug test.

MichaelAtOz, 2019-07-30:
@rcmaniac25 You may want to try [Googles version](https://fonts.google.com/specimen/Pathway+Gothic+One?selection.family=Pathway+Gothic+One).

It looks normal in the windows font list:
[image]

rcmaniac25, 2019-07-30:
I'll get to this in a bit, but how did you download the Google font? Unless there's a button I'm missing, those are generally linked for web usage

MichaelAtOz, 2019-07-30:
Can you try the --debug test on the original font & then the Google one. I deleted the original in Windows/fonts (right-click/delete).

To download it:
[image]
Click the Select Font, at the top, then the '1 Family Selected' 'window' appears at the bottom, click the white underscore on the right to slide it up, then click the right-curved red arrow. Intuitive...

rcmaniac25, 2019-07-30:
Sorry for the long delay.

Unblocking + clear cache + try and run: nothing changed
`-debug` output:
```
OpenSCAD 2019.07.28.ci3069
https://www.openscad.org/

Copyright (C) 2009-2019 The OpenSCAD Developers

This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Parsing design (AST generation)...
Saved backup file: E:/Documents/OpenSCAD/backups/unsaved-backup-mVSFUqJs.scad
Compiling design (CSG Tree generation)...
FreetypeRenderer: Detected script 'Latn' for hello world
FreetypeRenderer: Detected direction 'ltr' for hello world
Compiling design (CSG Products generation)...
FontCache: font = "Pathway Gothic One", lookup = "Pathway Gothic One"
FontCache: charmap = 0: platform = 0, encoding = 3
FontCache: charmap = 1: platform = 1, encoding = 0
FontCache: charmap = 2: platform = 3, encoding = 1
FontCache: Successfully selected unicode charmap: Liberation Sans/Regular
FontCache: result = "Liberation Sans", style = "Regular"
Geometries in cache: 1
Geometry cache size in bytes: 6736
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

renderer: setColorScheme
renderer: setColor b
renderer: setColor a
```

Will test the Google Version now and post again

rcmaniac25, 2019-07-30:
Google Font download: yep, that sure is "intuitive" but good to know as they have plenty of fonts that are useful

`debug` output:
```
OpenSCAD 2019.07.28.ci3069
https://www.openscad.org/

Copyright (C) 2009-2019 The OpenSCAD Developers

This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.

Parsing design (AST generation)...
Saved backup file: E:/Documents/OpenSCAD/backups/unsaved-backup-DTEzXXUr.scad
Compiling design (CSG Tree generation)...
FreetypeRenderer: Detected script 'Latn' for hello world
FreetypeRenderer: Detected direction 'ltr' for hello world
Compiling design (CSG Products generation)...
FontCache: font = "Pathway Gothic One", lookup = "Pathway Gothic One"
FontCache: charmap = 0: platform = 0, encoding = 3
FontCache: charmap = 1: platform = 1, encoding = 0
FontCache: charmap = 2: platform = 3, encoding = 1
FontCache: Successfully selected unicode charmap: Liberation Sans/Regular
FontCache: result = "Liberation Sans", style = "Regular"
Geometries in cache: 1
Geometry cache size in bytes: 6736
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
renderer: Renderer() start
renderer: setColorScheme
renderer: Renderer() end
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 1 seconds

renderer: setColorScheme
renderer: setColor b
renderer: setColor a
```

chris669, 2019-07-30:
hi Michael, I appreciate Openscad, great product, but you certainly know that Windows 7 EOL date is 15 January 2020, it could be an opportunity to get a W10 test station? I get my license on ebay!

MichaelAtOz, 2019-07-30:
[`PRINTDB("Successfully selected unicode charmap: %s/%s", face->family_name % face->style_name);`](https://github.com/openscad/openscad/blob/9e6e2ade126f339e6d89ca01eac30722cf53bc92/src/FontCache.cc#L377)
Means it selected the default, Liberation Sans/Regular

That is done by [FcFontMatch](https://github.com/openscad/openscad/blob/9e6e2ade126f339e6d89ca01eac30722cf53bc92/src/FontCache.cc#L353). 

...as it doesn't show up on the OpenSCAD font list that font is not getting loaded into the config. That is done by FcConfigAppFontAddDir. As all other fonts are loaded, it is unlikely this is an OpenSCAD code issue. 

So this is a Windows-10/Fontconfig-library bug, OR the font is misconfigured, or both...hmm, but the font works on Windows-7...

Unfortunately, it is going to require some debug code in there to see what the intermediate results are, I'm not a developer (at that level of detail).
So sorry, no quick fix for this ATM....

MichaelAtOz, 2019-07-30:
@chris669 I'm migrating to Debian, didn't like W10 becoming a marketing platform rather than an OS.

chris669, 2019-07-30:
@MichaelAtOz , sure, I totally agree with you, but I have so many apps that are not Linux compatible. But I was thinking maybe about a Virtual machine for testing? Anyway, tell me again if you need to test.

rcmaniac25, 2019-08-19:
@MichaelAtOz Can you do `set FC_DEBUG=9`? (I chose 9 just because it was large-ish) in CMD, then run OpenSCAD from CMD and pipe the logs to some file, then list fonts from OpenSCAD. Close OpenSCAD and post the file?

I'm trying to see if you get anything different from me. I've cross compiled OpenSCAD and (after another project) debugging what is going wrong with fonts.

MichaelAtOz, 2019-08-19:
@rcmaniac25 Note I have neither version of Pathway Gothic One loaded ATM. Something corrupted Windows Fonts, I has to go postal on Window to get rid of both as somehow they were both installed, but not showing in the Windows Fonts directory....(but one showed in OpenSCAD's list).

With `set FC_DEBUG=9`, `openscad > a.log` then `New`, then `Help/Font List`, crashes **openscad.com** (ie the CMD program that hands over to openscad.exe) just as the building font list dialogue shows, after clearing the crash dialogue the font list is shown. So I don't know whether any logged output is lost, but this is the result:

```
FC_DEBUG=9
adding fonts from c:/Program Files/OpenSCAD/fonts
adding fonts from c:/Program Files/OpenSCAD/fonts/Liberation-2.00.1
adding fonts from c:/Program Files/OpenSCAD/fonts/conf.d
adding fonts from c:/Program Files/OpenSCAD/fonts/Liberation-2.00.1/ttf
adding fonts from C:/Windows/fonts
adding fonts from C:/Users/MeB/.local/share/fonts
cache scan dir C:/Users/MeB/.local/share/fonts
adding fonts from C:/Users/MeB/.fonts
cache scan dir C:/Users/MeB/.fonts
FontSet 1049 of 1056
Font 0 Pattern has 23 elts (size 23)
	family: "Fixedsys"(w)
	style: "Regular"(w)
	stylelang: "en"(w) "en"(w)
	slant: 0(i)(w)
	weight: 80(f)(w)
	width: 100(f)(w)
	pixelsize: 16(f)(w)
	spacing: 100(i)(w)
	foundry: "unknown"(w)
	antialias: False(w)
	file: "C:/Windows/fonts/8514fix.fon"(w)
	index: 0(i)(w)
	outline: False(w)
	scalable: False(w)
	charset: 
(w)
	lang: (w)
	fontversion: 0(i)(w)
	fontformat: "Windows FNT"(w)
	decorative: False(w)
	postscriptname: "Fixedsys"(w)
	color: False(w)
	symbol: False(w
```

MichaelAtOz, 2019-08-19:
@rcmaniac25 I see it logs much more than that if I just let it log to the `CMD` window.
Unfortunately you can only set the screen buffer size (top-left-icon/preferences) to a max of 9999 lines.
Unfortunately I have **lots** of fonts and the first ~750 get dropped, and "path" (I reinstalled the goggle font) is not found in the remaining output.
Do you want the tail 9999 lines?

MichaelAtOz, 2019-08-19:
Maybe this is helpful, result using 
`text("hello world",font="Pathway Gothic One:style=Regular"); `

```

C:\Users\MeB>set FC_DEBUG=1

C:\Users\MeB>openscad
FC_DEBUG=1
Match Pattern has 27 elts (size 32)
        family: "Pathway Gothic One"(s) "Bitstream Vera Sans"(w) "DejaVu Sans"(w) "Verdana"(w) "Arial"(w) "Albany AMT"(w) "Luxi Sans"(w) "Nimbus Sans L"(w) "Nimbus Sans"(w) "Helvetica"(w) "Lucida Sans Unicode"(w) "B
PG Glaho International"(w) "Tahoma"(w) "Nachlieli"(w) "Lucida Sans Unicode"(w) "Yudit Unicode"(w) "Kerkis"(w) "ArmNet Helvetica"(w) "Artsounk"(w) "BPG UTF8 M"(w) "Waree"(w) "Loma"(w) "Garuda"(w) "Umpush"(w) "Saysett
ha Unicode"(w) "JG Lao Old Arial"(w) "GF Zemen Unicode"(w) "Pigiarniq"(w) "B Davat"(w) "B Compset"(w) "Kacst-Qr"(w) "Urdu Nastaliq Unicode"(w) "Raghindi"(w) "Mukti Narrow"(w) "malayalam"(w) "Sampige"(w) "padmaa"(w)
"Hapax Berb├¿re"(w) "MS Gothic"(w) "UmePlus P Gothic"(w) "Microsoft YaHei"(w) "Microsoft JhengHei"(w) "WenQuanYi Zen Hei"(w) "WenQuanYi Bitmap Song"(w) "AR PL ShanHeiSun Uni"(w) "AR PL New Sung"(w) "MgOpen Modata"(w
) "VL Gothic"(w) "IPAMonaGothic"(w) "IPAGothic"(w) "Sazanami Gothic"(w) "Kochi Gothic"(w) "AR PL KaitiM GB"(w) "AR PL KaitiM Big5"(w) "AR PL ShanHeiSun Uni"(w) "AR PL SungtiL GB"(w) "AR PL Mingti2L Big5"(w) "´╝¡´╝│
Òé┤ÒéÀÒââÒé»"(w) "ZYSong18030"(w) "TSCu_Paranar"(w) "NanumGothic"(w) "UnDotum"(w) "Baekmuk Dotum"(w) "Baekmuk Gulim"(w) "KacstQura"(w) "Lohit Bengali"(w) "Lohit Gujarati"(w) "Lohit Hindi"(w) "Lohit Marathi"(w) "Lohi
t Maithili"(w) "Lohit Kashmiri"(w) "Lohit Konkani"(w) "Lohit Nepali"(w) "Lohit Sindhi"(w) "Lohit Punjabi"(w) "Lohit Tamil"(w) "Meera"(w) "Lohit Malayalam"(w) "Lohit Kannada"(w) "Lohit Telugu"(w) "Lohit Oriya"(w) "LK
LUG"(w) "FreeSans"(w) "Arial Unicode MS"(w) "Arial Unicode"(w) "Code2000"(w) "Code2001"(w) "Liberation Sans"(s) "sans-serif"(w) "Roya"(w) "Koodak"(w) "Terafik"(w)
        familylang: "en"(s) "en-us"(w)
        style: "Regular"(s)
        stylelang: "en"(s) "en-us"(w)
        fullnamelang: "en"(s) "en-us"(w)
        slant: 0(i)(s)
        weight: 80(i)(s)
        width: 100(i)(s)
        size: 12(f)(s)
        pixelsize: 12.5(f)(s)
        hintstyle: 1(i)(w)
        hinting: True(s)
        verticallayout: False(s)
        autohint: False(s)
        globaladvance: True(s)
        outline: True(s)
        scalable: True(s)
        dpi: 75(f)(s)
        scale: 1(f)(s)
        lang: "en"(w)
        fontversion: 2147483647(i)(s)
        embeddedbitmap: True(s)
        decorative: False(s)
        namelang: "en"(s)
        prgname: "openscad.exe"(s)
        symbol: False(s)
        variable: False(s)

Best score 0 0 0 0 0 0 0 0 0 0 1001 0 0 0 0 0 0 0 0 0 0 0 0 0 2.14742e+12
Pattern has 24 elts (size 24)
        family: "Pathway Gothic One"(w)
        familylang: "en"(w)
        style: "Regular"(w)
        stylelang: "en"(w)
        fullname: "Pathway Gothic One Regular"(w)
        fullnamelang: "en"(w)
        slant: 0(i)(w)
        weight: 80(f)(w)
        width: 100(f)(w)
        foundry: "TIPO"(w)
        file: "C:/Windows/fonts/PathwayGothicOne-Regular.ttf"(w)
        index: 0(i)(w)
        outline: True(w)
        scalable: True(w)
        charset:
        0000: 00000000 ffffffff ffffffff 7fffffff 00000000 ffffdfef ffffffff ffffffff
        0001: 00000000 00020000 000c0006 61000003 00040000 00000000 00000000 00000000
        0002: 00000000 00000000 00000000 00000000 00000000 00000000 3f0000c0 00000000
        0003: 00000000 00000000 00000000 00000000 00000000 00000000 00000001 00000000
        0020: 77180000 06010047 00000010 00000000 000003ff 00001000 00000000 00000000
        0021: 00000000 00000044 00000000 00000000 00000000 00000000 00000000 00000000
        0022: 44068044 00000800 00000100 00000031 00000000 00000000 00000000 00000000
        0025: 00000000 00000000 00000000 00000000 00000000 00000000 00000400 00000000
        00fb: 00000006 00000000 00000000 00000000 00000000 00000000 00000000 00000000
(w)
        lang: aa|ay|bi|br|ch|co|da|de|en|es|et|eu|fi|fj|fo|fr|fur|fy|gd|gl|gv|ho|ia|id|ie|io|is|it|lb|mg|nb|nds|nl|nn|no|nr|nso|oc|om|pt|rm|sma|smj|so|sq|ss|st|sv|sw|tl|tn|ts|uz|vo|vot|wa|xh|yap|zu|an|fil|ht|jv|kj|k
wm|li|ms|ng|pap-an|pap-aw|rn|rw|sc|sg|sn|su|za(w)
        fontversion: 65667(i)(w)
        capability: "otlayout:latn"(w)
        fontformat: "TrueType"(w)
        decorative: False(w)
        postscriptname: "PathwayGothicOne-Regular"(w)
        color: False(w)
        symbol: False(w)
        variable: False(w)
```

rcmaniac25, 2019-08-20:
@MichaelAtOz Thanks. I had basically identical output (though piping the output, I achieved over 40k lines of logs about the fonts alone)

I wanted to search for the shortened name that is often stored in the fonts folder and... found the issue. My font wasn't in the Windows Font folder.

For more info: https://gitlab.freedesktop.org/fontconfig/fontconfig/issues/144

I worked around this by adding the following to the `fonts.conf` file:
```
<dir>~/AppData/Local/Microsoft/Windows/Fonts</dir>
<dir>~/AppData/Roaming/Microsoft/Windows/Fonts</dir>
```

With that... success! Now, I doubt it's a preferred solution, but until fontconfig adds a solution, I don't think there's a solution without a tmp hack or a config change. Config change is probably easier in the short term.

MichaelAtOz, 2019-08-21:
For any W10 users, you can get fonts installed to the 'proper' location, see:
https://stackoverflow.com/questions/55264642/how-to-force-win10-to-install-fonts-in-c-windows-fonts

This is due to a change in Windows 10, for non-admin users installing fonts they get installed to AppData (see above). The link above is a manual work-around.

@t-paul Should fonts.conf (or another .conf)be updated for the upcoming Patch release, until it is fixed in fontconfig?

As a side note, this is from the FC_DEBUG output:
```
FC_DEBUG=9
adding fonts from c:/Program Files/OpenSCAD/fonts
adding fonts from c:/Program Files/OpenSCAD/fonts/Liberation-2.00.1
adding fonts from c:/Program Files/OpenSCAD/fonts/conf.d
adding fonts from c:/Program Files/OpenSCAD/fonts/Liberation-2.00.1/ttf
adding fonts from C:/Windows/fonts
adding fonts from C:/Users/MeB/.local/share/fonts
cache scan dir C:/Users/MeB/.local/share/fonts
adding fonts from C:/Users/MeB/.fonts
cache scan dir C:/Users/MeB/.fonts
FontSet 1050 of 1056
Font 0 Pattern has 23 elts (size 23)
	family: "Fixedsys"(w)
...
```

`.fonts` is [here](https://github.com/openscad/openscad/blob/b58875721e2c98ac4b9bbb31d0968671beb507e7/src/FontCache.cc#L148) perhaps it should be skipped for Windows? 

 I couldn't find `.local/...` Is that just fontconfig being Linux parochial?

rcmaniac25, 2019-08-21:
`./local/share` is the done as the default for the `xdg` in `<dir prefix="xdg">fonts</dir>` (src: xdg is substituted with this path: https://gitlab.freedesktop.org/fontconfig/fontconfig/blob/master/src/fccfg.c#L2251)

closed 2026-03-10 by github-actions[bot]