openscad/openscad #2513: Warnings of variables overridden in used files need to show the file name.

nophead, 2018-10-08:
```
WARNING: length was assigned on line 25 but was overwritten on line 34 
WARNING: length was assigned on line 72 but was overwritten on line 79 
```
When the lines are not in the main file the filename should be mentioned, otherwise it can be hard to find in a library of hundreds of files. This is especially true since this is a new warning so it appears in old projects, so likely isn't a file that has recently changed.

MichaelPFrey, 2018-10-09:
Thank you for the report.

I have a working patch that disables the duplicate assignment warning for "used" files.
The idea is a following: The user might "use" libraries that he has no interest in modifying and/or that a legacy code, so making warnings for code that is isolated from the current code is counter productive.
A library maintainer gets the warning, when he or she is editing the code of the library it self.

Even if one considers the warning to be useful "anyway": I can not create a nice relative path. Outputting "sometimes" just filenames and sometimes relative paths  (that are not always relative to the file currently being edited) with filename would be confusing.

nophead, 2018-10-10:
>A library maintainer gets the warning, when he or she is editing the code of the library it self.

Not in my case because I don't ever load the used file into OpenSCAD directly. It is always used by another file that instantiates the module, either a project or a test file. The way my library is structured object descriptions are lists that are included and the code to draw them is used. Using it on its own doesn't draw anything because the descriptions are in another file that uses it.

I think OpenSCAD can always get the full path of a file it uses so we could output that or also get the full path of the main file and then use that to calculate the minimum relative  path.

MichaelPFrey, 2018-10-10:
> Using it on its own doesn't draw anything because the descriptions are in another file that uses it.
I tend to write some short self testing code in libraries I want to use using use.
The self-test getting ignored when beeing used.

> I think OpenSCAD can always get the full path of a file it uses, so we could output that or also get the full path of the main file and then use that to calculate the minimum relative path.

In theory: Sure. Give me two paths and I can use boostfs_uncomplete to get a relative path.

In practice: Tricky.

One example on why it is tricky: We currently do not properly canonicalize the paths in the test framework, making the testframe work unable to truly test boosts uncomplete function in context of assert and duplicate assignments.

More specifically with used files: Currently, the parser only know the RootFile, not the mainFile (why should it? It does not need it currently)

We could of course change the datastructur, but I do not fully understand the datastructures involved. For e.g. we have a module cache and the user can have multiple openscad windows open.

A simple test showing the complexity involved: Make a syntax error in an "used" scad file: The file beeing "used" is only parsed once, so you only get the message for the error in the used scad file only for the first compile main file (or when you change the used file and then recompile the main file).

Thinking about it: Syntax error show an interesting handling of errors: It simply shows the full path of the file in the file that was being parsed. That would work - except for the regression test frame work.

We could also simply output the file name without path - that would work for regression testing, but I think we can agree that the path is very useful (if not essential) for finding the offending file.

The trick with assert and use/include in getting the paths was to exploit the exception handling. (which works for assert that causes an exception, but not for duplicate assignments that are just warnings)

I mean: If someone knows what he or she is doing: Go a head and improve OpenSCADs error reporting. So, I leave this issue open for now.

In the mean time, suppressing the warning as in #2516 seams reasonable.

nophead, 2018-10-10:
>We currently do not properly canonicalize the paths in the test framework, making the testframe work unable to truly test boosts uncomplete function in context of assert and duplicate assignments.

Are you saying the problem is if the test framework outputs a full path it will be different on different people's machines because it depends where they build it? 

If that is the problem then I think outputting a relative path solves it because the non-canonical bit of the path disappears.  The test files always have the same relationship between themselves.

kintel, 2018-10-13:
I think nophead concern with this can be extended to included files.
None of the following two scenarios currently trigger the warning, but they probably should:

main.scad:
```
include <included.scad>
```

included.scad:
```
overwrittenInInclude = false;
overwrittenInInclude = true;
```

main2.scad:
```
include <included2.scad>
before=false;
```

included2.scad:
```
before=true;
```

MichaelPFrey, 2018-10-13:
included.scad and main.scad: Oh, that is an easy change, if you want a more verbose warning system. The first implementation was carefully conservative to only warn, when there is "no doubt about it" an issue with the duplicate assignment, that directly affects the main file.

included2.scad and main2.scad intentionally does not trigger a warning. The use case for this is to use inlcuded2.scad as a container for default values, that can be overwritten by other code. Note that included2.scad can be included into many files (e.g. main2a.scad, main2b.scad, ...)

As a rough rule of thumb I use "visibility" and "could the assignment have a meaning in some context". 

Visibility meaning: Does the an assignment that is more visible to the user take priority? If not, do warn.
(e.g. and included file overwriting an assignment of the main file)

"Could the assignment have a meaning in some context" is shown in included2.scad and main2.scad: included2.scad could be used for multiple files, so before=true; could be a default. (or included2.scad could be a full part and main2.scad an assembly, a variant of the part, ...) 

I mean, there are multiple ways of handling duplicate assignments:
- outright error duplicate assignments
   - breaks legacy code
   - breaks the command line variable overwrite
   - breaks legitimate use case (included2.scad being a container for default values, intended to be overwritten "as needed")
- warn about duplicate assignments
- inform about duplicate assignments
- silently use the last assignment
   - (which openscad has done for the longest time)

Adding a new reporting level "information" would be an elegant way of handling included2.scad and main2.scad. Information would also work for command line variable overwrite. That sounds right to me - always inform the user of duplicate assignments. Warn, when the overwritten assignment can not take effect under any circumstances.

What I did not take into account is, when the assignments are not using constants. If we going serious, we also have to address this:
```
a=1;
b=2;
a=b*2;
```
> WARNING: Ignoring unknown variable 'b'. 

in all variations with include.

jordanbrown0, 2025-03-09:
The original request was to include the file name in the warning message.  That appears to have been done:

```
WARNING: "x" was assigned on line 2 of "b.scad" but was overwritten in file b.scad, line 3
```

closed 2025-03-09 by jordanbrown0