openscad/openscad #2514: clean up the experimental options in tests/CMakeLists.txt

MichaelPFrey, 2018-10-09:
As a general note: due to #2499 we should remove experimental options that have become stable in the mean time:
https://github.com/openscad/openscad/blob/master/tests/CMakeLists.txt#L1032 

features that became most recently stable: #2505 #2504 

So fare I think we need to remove `"--enable=assert" "--enable=echo" "--enable=lc-each"` from https://github.com/openscad/openscad/blob/master/tests/CMakeLists.txt#L1032

t-paul, 2018-10-09:
The option itself (which is just `--enable`) is not going away. It will always be available with `CONFIG+=experimental`. So there should be no issue with #2499.

MichaelPFrey, 2018-10-09:
@t-paul you are correct, there is no link to #2499 - I confused my self by not building openscad with the experimental option `CONFIG+=experimental.` I will update the issue accordingly. Cleaning up the option still makes sense tough.

t-paul, 2018-10-09:
Sure, cleanup is already in progress :-).

commit 832ad3c6 (2018-10-09): Remove experimental flags for assert and echo (fixes #2514).
commit f3de0e27 (2018-10-09): Merge pull request #2515 from openscad/test-cleanup

Remove experimental flags for assert and echo (fixes #2514).
closed 2018-10-09 by t-paul