openscad/openscad #2517: DEPRECATED: Imported file (blah.stl) found in document root instead of relative to the importing module. This behavior is deprecated

HarlemSquirrel, 2018-10-11:
```
import("blah.stl");
```

The above results in this warning:

>DEPRECATED: Imported file (blah.stl) found in document root instead of relative to the importing module. This behavior is deprecated

I'm not sure what "relative to the importing module" refers to. 

Providing an absolute path gets rid of the deprecation warning but this is not great since I could move the project folder. I do not see anything on this in [the wiki](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Import_and_Export).

nophead, 2018-10-11:
It means the path in the import statement should be relative to the file with the the import statement in it, not the main file you have open in the OpenSCAD window. Is your import in a module in a different directory to the main file?

HarlemSquirrel, 2018-10-11:
I'm not importing or including any other OpenSCAD files. The STL file is in
the same folder as the OpenSCAD file.

I'm using ArchLinux by the way.

Kevin McCormack
Full Stack Web Developer
HarlemSquirrel.github.io

MichaelPFrey, 2018-10-11:
which OpenSCAD version are you using?

HarlemSquirrel, 2018-10-11:
I'm using 2015.03.03-18 which is the latest version in the [ArchLinux community repo](https://www.archlinux.org/packages/community/x86_64/openscad/).

t-paul, 2018-10-12:
See quick description in https://github.com/openscad/openscad/issues/217#issuecomment-17575668 - I've also added a note to the manual.

closed 2018-10-12 by t-paul