openscad/openscad #2518: Sync top level cmake setup with qmake build.

t-paul, 2018-10-13:
The qmake build is still the official one, but we do want to move to full cmake build eventually. So this issue is here as reminder to re-sync.

- [ ] Update for the changes the inputdriver branch brought in

thehans, 2019-11-15:
I checked off a couple of the boxes here, 

Input driver stuff also works afaik, but I don't use any special input devices to test with.
I'm not sure what the `QT5QSCINTILLA_LIBRARY` issue is about, but has been working fine for me.
`HEADLESS` option was also working last time I checked.

I think maybe the only one still relevant is to add install targets.

t-paul, 2022-01-19:
Closing, snapshot builds are `cmake` only for a while.

closed 2022-01-19 by t-paul