openscad/openscad #2527: Ignore Transparent sections of SVGs

celesteh, 2018-10-21:
It would be nice if that when using SVGs for linear_extrude and the like that the transparent portions were ignored. This aligns more closely with what the user would expect to see.

This may be related to issue #2145

MichaelPFrey, 2018-10-21:
#2145 read like an issue with exporting svg.
To me, this reads like an issue with importing svg. 
Please correct me if I am wrong.

t-paul, 2018-10-21:
I'm not sure it's viable to look too much at the graphics attributes. What do we do if there's a gradient? The problem is that SVG is a graphics and not a CAD format, and that part will probably never import perfectly.

That said, are there any links to example files (preferably things like OpenClipart which could be used and distributed as test cases). Maybe there are some cases that are reasonable to cover.

celesteh, 2018-10-22:
Sorry, I misread #2145 and have edited the original post to remove mention of it.

t-paul, 2019-11-18:
Closing as out-of-scope now as there was no further feedback. We might revisit at some point, but as mentioned earlier there's a mismatch between graphical/visual properties of SVG and the need of OpenSCAD to import geometric structures.

closed 2019-11-18 by t-paul