openscad/openscad #2528: Wrong shapes for some SVG paths

celesteh, 2018-10-21:
Some SVG paths are traced to incorrect shapes.

This SVG produces an incorrect output:

```
<svg id="Logo_FIXED" data-name="Logo — FIXED" xmlns="http://www.w3.org/2000/svg" viewBox="0 0 400 400"><defs><style>.cls-1{fill:none;}.cls-2{fill:#1da1f2;}</style></defs><title>Twitter_Logo_Blue</title>
<path class="cls-2" d="M153.62,301.59c94.34,0,145.94-78.16,145.94-145.94,0-2.22,0-4.43-.15-6.63A104.36,104.36,0,0,0,325,122.47a102.38,102.38,0,0,1-29.46,8.07,51.47,51.47,0,0,0,22.55-28.37,102.79,102.79,0,0,1-32.57,12.45,51.34,51.34,0,0,0-87.41,46.78A145.62,145.62,0,0,1,92.4,107.81a51.33,51.33,0,0,0,15.88,68.47A50.91,50.91,0,0,1,85,169.86c0,.21,0,.43,0,.65a51.31,51.31,0,0,0,41.15,50.28,51.21,51.21,0,0,1-23.16.88,51.35,51.35,0,0,0,47.92,35.62,102.92,102.92,0,0,1-63.7,22A104.41,104.41,0,0,1,75,278.55a145.21,145.21,0,0,0,78.62,23"/></svg>
```

The workaround is to simplify the path in inkscape, which produces minor distortions into the shape. All graphics programs seem to render the original path correctly, but the svg2stl website also struggles with it, although it produces a different wrong shape.

MichaelPFrey, 2018-10-21:
Above OpenSCAD, below FireFox. I think we can assume the firefox rendering to be correct for the purpose of this issue.
[image]

MichaelPFrey, 2018-10-21:
getting closer by reducing the path to smaller failing example:
[image]
```
<svg id="Logo_FIXED" data-name="Logo — FIXED" xmlns="http://www.w3.org/2000/svg" viewBox="0 0 400 400">
<path style="fill:#1da1f2;" d="
M153.62,301.59
A50.91,50.91,0,0,1,85,169.86
a51.31,51.31,0,0,0,41.15,50.28,51.21,51.21,0,0,1-23.16.88,51.35,51.35,0,0,0,47.92,35.62,102.92,102.92,0,0,1-63.7,22
A104.41,104.41,0,0,1,75,278.55
a145.21,145.21,0,0,0,78.62,23
"/></svg>
```

t-paul, 2018-10-21:
Nice, thanks for the example, it looks like it breaks with the sweep flag.
```
M153.62,301.59
c94.34,0,145.94-78.16,145.94-145.94,
 0-2.22,0-4.43-.15-6.63
A104.36,104.36,0,0,0,325,122.47
a102.38,102.38,0,0,1-29.46,8.07,
 51.47,51.47,0,0,0,22.55-28.37,
 102.79,102.79,0,0,1-32.57,12.45,
 51.34,51.34,0,0,0-87.41,46.78
A145.62,145.62,0,0,1,92.4,107.81
a51.33,51.33,0,0,0,15.88,68.47
A50.91,50.91,0,0,1,85,169.86
c0,.21,0,.43,0,.65
a51.31,51.31,0,0,0,41.15,50.28,
 51.21,51.21,0,0,1-23.16.88,     <-- Sweep flag set
 51.35,51.35,0,0,0,47.92,35.62,
 102.92,102.92,0,0,1-63.7,22
A104.41,104.41,0,0,1,75,278.55
a145.21,145.21,0,0,0,78.62,23
```

commit ea92356a (2018-10-25): Add workaround for number parsing (fixes #2528).
closed 2018-10-25 by kintel