openscad/openscad #2536: Editor/console window width resets frequently

maaltan, 2018-10-26:
i just upgraded to OpenSCAD version 2018.10.13.ci145 (git 022f50e9)  and have run into an extremely annoying bug.  I am using the portable/zip package if that makes a different.

It seems like every time the render window updates, the window layout snaps back to default/starting width (which makes the editor about 30 characters wide on my screen).  

Here are some specific tasks that cause this:
* refresh after a change to geometry (ie, it has to redraw the object), or compile error
* rotate/zoom/pan/etc in preview window.
* resizing application window.

It has been a while since i updated(2015.3-2) , but the window layout persisted even through restarts then.

t-paul, 2018-10-26:
This is probably caused by this Qt bug: https://bugreports.qt.io/browse/QTBUG-65592

t-paul, 2018-12-08:
I still can't reproduces this here but I've added the workaround suggested in the ticket linked above. Can you please have a look if that fixes the issue?

64bit: https://circleci.com/gh/openscad/openscad/706#artifacts
32bit: https://circleci.com/gh/openscad/openscad/707#artifacts

t-paul, 2019-01-25:
Closing as this should be fixed via #2627.

closed 2019-01-25 by t-paul