openscad/openscad #2548: Customizer and Animations

MichaelPFrey, 2018-11-03:
Customizer and Animations do not play nicely with each other.

Part of the issue is, that "traditionally", you can change the code while an animation is running.
To support that from the customizers point of view, we need to rebuild the customizer after each change of the code. As each animation cycle could include changed code, resulting in the pointers of customizer pointing to outdated datastructures, we have to "in doubt" rebuild the customizer.

Oneway to resolve that would be to make two animation interfaces. One being the current one. The other being integrated in to the customizer. But: Having two animation interfaces is confusing and likely to cause other conflicts.

The better way to go would be that the animation controls get an option to either lock the code or the customizer. If the customizer is locked during animation, the animations behaves like it does now.
If the code is locked, the customizer is enabled and the animation now affects the design via the customizer.

From that - the user choosing whether the Code or the customizer is locked - we could make that automatically. So, if the user double clicks on the code, the customizer is locked and the code unlocked. If the user double clicks the customizer, the code is locked and the customizer unlocked. 

Maybe someone has a better idea and wants to give it a try.

This is part of #2316, but also https://github.com/openscad/openscad/issues/2378#issuecomment-435438264

MichaelPFrey, 2018-11-04:
I tried to make a bodge as prove of concept, but I got stuck. The customizer is doing something strange with the focus.

It might be an idea to stop the animation while the customizer has focus. That would be fairly simple and could be made to feel consistent.

Philius, 2018-11-07:
[hypocycloid.scad.gz](https://github.com/openscad/openscad/files/2556650/hypocycloid.scad.gz)
The model doesn't render when OpenSCAD loads it until you press reset in the customizer but it does show in animation! I chose frame rate = 30 steps = 30.