openscad/openscad #2575: create warning when vector pased to inbuilt module is invalid

MichaelPFrey, 2018-11-12:
Two examples:
`translate([10, 0, "ohno"]) cube([2,2]);`
Note that sometimes we want to default the second parameter.

To find such cases, the code can be searched for getVec3 and getVec2.

MichaelPFrey, 2018-11-13:
making progress:
[image]
offcourse beeing pased on  #2543, without line numbers it would be really tricky.
Regression testing will be tricky due to  #2566.

Note that not all getVec calls should get warnings. This one for e.g. belongs to an deprecated feature.
https://github.com/openscad/openscad/blob/6670a636bcad0e2f6db0ead24e7e3707eab56708/src/rotateextrude.cc#L82

MichaelAtOz, 2018-11-13:
`cube();` and `sphere();` etc which default to 1, are commonly used in testing. I doubt it is in usable code tho.

Is there any practical difference between a WARNING and an ERROR? The above looks like ERROR still compiles objects.

nophead, 2018-11-13:
cube(n) is supposed to work is not? And I thought the default value was 1 so cube(); should be valid.

MichaelPFrey, 2018-11-14:
> Is there any practical difference between a WARNING and an ERROR? 

Opps... yes they should be warnings, not erros. Thank you, I will fix that.

> cube(n) is supposed to work is not? 

Yes, I over looked, that, but that should be easy to fix.

> And I thought the default value was 1 so cube(); should be valid.

Very good input. I will look into on how I can address that.

MichaelPFrey, 2018-11-14:
interestingly `cube();` is actually `cube(undef);` not as one might expect `cube(1);`, but `cube(undef);` also has unity size.

[image]

thehans, 2018-11-17:
@MichaelPFrey since you are looking into this, and it seems somewhat related to #1646, do you think that could also be addressed in these changes?
In summary: when a scalar is passed to scale, it puts the scalar in all 3 components, even if the geometry being scaled is 2D, which breaks the preview.

MichaelPFrey, 2018-11-17:
@thehans Tough question. I see two problems: At the point where I am implementing the checks, I have no geometry information. So, I do not know whether the scale belongs to a 2D or 3D Objects. The second problem - coming into play when we have access to the geometry - is mixed 2d/3d objects:
```
scale(1/3) {
     square(center=true);
    translate([2,0])
    cube(center=true);
}
```
Yes, this is nonsense, but somewhere somehow  we need to handle that.
Tricky. Currently I do not know where to look for the relevant code. I will keep it in the back of my mind, maybe I stumble over the solution when I am looking for something else.

MichaelPFrey, 2019-02-03:
some cases to think about:

- [ ] `mirror([0,0,0])cube();` ("if in doubt" that is fine)
- [ ]  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Mathematical_Functions#lookup
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/func.cc#L571
- [ ]  `import("test.dxf", origin="test");`
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/dxfdim.cc#L67
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/dxfdim.cc#L156
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/import.cc#L129
- [ ]  linearextrude -> origin, scale
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/linearextrude.cc#L99
https://github.com/openscad/openscad/blob/d8a6ef32a590904478c6a4cf6a7346e6d7cf7d1d/src/linearextrude.cc#L100-L103
- [ ]  rotateextrude -> origin

Okay - still some things, mostly related to getVec2.

closed 2026-03-12 by github-actions[bot]