openscad/openscad #2581: Cube with zero dimension should become equivalent to square.

ioquatix, 2018-11-13:
It would be nice if a cube with a zero dimension could still be useful as part of "hull" operations. This simplifies parametric objects where you can specify the thickness, which is part of a hull, right now if you specify 0 for any dimension, the shape is not considered as part of the hull operation.

t-paul, 2018-11-13:
There should be a feature that can achieve that goal. But it's not really possible by using a degenerated cube as 2D and 3D objects live in strictly distinct dinstinct worlds with some defined ways to cross over (like extrusion).

ioquatix, 2018-11-13:
It should be possible to make a hull out of two 2D objects or a 2D and 3D object, no?

nophead, 2018-11-13:
Yes hull only needs vertices, so it should be possible.

ioquatix, 2018-11-13:
It would be nice if degenerate 3D objects could take part in hull operations.

ioquatix, 2018-11-13:
Right now I just set it to a very small non-zero number, but that bothers me a little bit :p

MichaelPFrey, 2018-11-14:
I am always a bit concerned about consistency of the language. When a user is only uses 3D primitives in  the code and uses an array to feed the code, it would get confusing when suddenly openscad starts warning about 2D objects, just because somewhere, there is an [0,1,3] in the array.

What you could do: You can always over write built in modules or implement your own module. So something like:
```
module box(a){
    if(a[0]>0 && a[1]>0 && a[2]>0){
        cube(a);
    }
    if(a[0]>0 && a[1]>0){
        cube([a[0],a[1],0.0001]);
    }
    if(a[1]>0 && a[2]>0){
        cube([0.0001,a[1],a[2],]);
    }
    if(a[0]>0 && a[2]>0){
        cube([a[0],0.0001,a[2],]);
    }
}

box([1,2,3]);
box([5,4,0]);
box([3,0,6]);
```
Off course, you can use a true square when a parameter is zero.
You can also call that module cube, if you reimplement cube, using for e.g. polyhedron.

t-paul, 2019-11-14:
As there are other open requests for 2D support (e.g. #99) and no plan to support degenerate geometry as proposed, I'm closing this issue.

closed 2019-11-14 by t-paul

nophead, 2019-11-14:
I have a lot of modules that make holes that I sometimes use in 3D differences and often in 2D differences.  To make 3D shapes degenerate into 2D (avoiding projection() which is very slow) I use a module called extrude_if(h) which extrudes if h is non zero and returns a 3D object but if h is zero it returns the 2D original shape. https://github.com/nophead/NopSCADlib/blob/8583f5472d46ba231887ff02e8e267111cafd746/utils/core/global.scad#L44