openscad/openscad #2584: Valid mesh causes incorrect projection/intersection.

ioquatix, 2018-11-14:
This file renders the correct projection in the latest stable release but fails in 2018 with no output.

https://github.com/ioquatix/matx-case/blob/2bc868420cd77dcbabd28d13329d29430cbeaf23/source/projections.scad

The validity of the mesh is fine.

I tried doing an intersection instead of a projection, and it seemed to produce inverted output.

Additionally, trying to "check validity" of that resulting mesh caused OpenSCAD to crash completely.

kintel, 2024-03-30:
Status as of 2024-03-24: This seems to work now with CGAL, but not with Manifold

ioquatix, 2024-04-01:
Anything I can do to help?

kintel, 2024-04-01:
If you could verify that this is fixed in the latest development snapshot that would be great!

ioquatix, 2024-04-03:
As far as I can tell, this now appears to be working.

kintel, 2024-04-03:
Reopening to track this Manifold issue. Changed title and edited description to reflect this.

ioquatix, 2024-04-03:
Maybe you can test using this more up to date file: https://github.com/ioquatix/matx-case/blob/main/parts/panels.scad

ioquatix, 2024-04-03:
Also, somewhat related but not really, https://github.com/ioquatix/matx-case/commit/50a230f0a75b9b7d957669fbb847c6cdbc65b01f was required to get my code to render with the latest version of OpenSCAD - any thoughts on how to achieve this in a compatible way? Sometimes I used this so I could set the variable in a different file... I've just set it to use a constant for now.

kintel, 2024-04-03:
Could you open a separate issue with the $variable stuff? There were some similar discussions going on, and we need a good use-case to resurface them.

ioquatix, 2024-04-03:
Okay, done: https://github.com/openscad/openscad/issues/5075

kintel, 2024-04-03:
PS! I got inspired to pick up an optimization opportunity: With this PR, your panels.scad model renders in 1 second on my machine: https://github.com/openscad/openscad/pull/5076

Not sure yet if it renders correctly, but it looks reasonable:
[image]

ioquatix, 2024-04-03:
That looks correct to me.

Speeding up the performance would be awesome!

I would like to adopt this style of design for laser cutting more often.

Therefore, efforts to improve this would be most welcome.

PR #5076 (2024-04-08): Use Manifold's Slice() and Project() to implement projection()
This should give `projection(cut=true)` a few orders of magniture speed boost.

Also fixes #2584
closed 2024-04-08 by kintel