openscad/openscad #2590: Trimming cache output... bug or feature?

boxcarmib, 2018-11-16:
Downloaded and installed Mac OpenSCAD version 2018.11.05 (git e1f043d2) and with this release rendering dumps a whole bunch of Trimming cache output to the console window. I've looked at preferences as there doesn't appear to be any parameter that can disable or inhibit this behaviour and for me anyway... it isn't particularly helpful output. Am I missing something?

Trimming cache: group(){difference(){group(){group(){polyhedron(points=[[0,9.95,0.05],[0.624766,9.93037,0.05],[1.24707,9.87154,0.05],[1.86444,9.77376,0.05],[2.47446,9.6374,0.05],[3.07472,9.46301,0.05],[3.66284,9.25128,0.05],[4.2365,9.00303,0.05],[4.79345,8.71925,0.05],[5.33148,8.40106,0.05],[5.84846,8.04972,0.05],[6.34237,7.66661,0.05],[6.81124,7.25324,0.05],[7.25324,6.81124,0.05],[7.66661,6.34237,0.05],[8.04972,5.84846,0.05],[8.40106,5.33148,0.05],[8.71925,4.79345,0.05],[9.00303,4.2365,0.05],[9.25128,3.66284,0.05],[9.46301,3.07472,0.05],[9.6374,2.47446,0.05],[9.77376,1.86444,0.05],[9.87154,1.24707,0.05],[9.93037,0.624766,0.05],[9.95,0,0.05],[9.93037,-0.624766,0.05],[9.87154,-1.24707,0.05],[9.77376,-1.86444,0.05],[9.6374,-2.47446,0.05],[9.46301,-3.07472,0.05],[9.25128,-3.66284,0.05], ........

MichaelAtOz, 2018-11-19:
@boxcarmib While not a bug as such, and also a feature (debug setting) it probably should not be set in the snapshot. I believe Mac snapshots are built differently, so it could be an oversight.
I would expect it to disappear in future snapshots. @kintel?

kintel, 2018-11-19:
I'll have a look at the build settings.

kintel, 2018-11-25:
This was a local configuration mistake on my Mac build machine. Fixed in OpenSCAD-2018.11.25

closed 2018-11-25 by kintel