openscad/openscad #2597: is_undef() missing from Mac version 2018.11.05 (git e1f043d)

boxcarmib, 2018-11-20:
just going by the current documentation...
for  Mac OpenSCAD version 2018.11.05 (git e1f043d)
the following code:

echo(is_undef($exploded));

returns

Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown function 'is_undef'. 
ECHO: undef
Compiling design (CSG Products generation)...
Geometries in cache: 73
Geometry cache size in bytes: 549512
CGAL Polyhedrons in cache: 11
CGAL cache size in bytes: 3460832
Compiling design (CSG Products normalization)...
Normalized CSG tree has 0 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds

t-paul, 2018-11-20:
This was merged on 2018-11-11 so it will be included in the next snapshot build.

closed 2018-11-20 by t-paul

t-paul, 2018-11-26:
There's a new macOS snapshot available now.