openscad/openscad #2601: Trigger "View All" on program start

korpx, 2018-11-23:
It'd be very nice if there was a checkbox in the Preferences to make "View All" trigger when OpenSCAD starts.

I find myself using View All (ctrl + shift + V) every time I open an OpenSCAD file to get everything into view. I find this more natural than centering on 0,0,0.

t-paul, 2018-11-23:
What you are asking is not possible as "when OpenSCAD starts" there is no geometry to center of.

I think what you mean is to center on first preview or render? That might be possible, but those kind of automatic delayed actions tend to get messy soon (e.g. what if the script changes viewport via `$vpX` variables).

korpx, 2018-11-23:
Yes, I mean when OpenSCAD load a geometry.
(E. g when OpenSCAD starts by file system association *.scad or starting OpenSCAD, File, Open....)

korpx, 2018-11-23:
But not on automatic reload (from using Automatic Reload and Preview) as that would surely drive anyone insane. Hopefully File, Open... and an automatic reload can be handled separately,

t-paul, 2018-11-23:
That's what I mean, if the file is loaded, there is no information about the geometry yet. This requires either a preview (F5) or a render operation (F6). Display on file load only happens with "automatic reload and preview" activated.

korpx, 2019-02-21:
From what I understand the effort in solving this request is unproportional to the experienced problem.

closed 2019-02-21 by korpx