openscad/openscad #2603: New windows are very tiny on high-DPI monitors on Windows

hrydgard, 2018-11-24:
When opening files or just starting the program on a high-dpi monitor, windows are created tiny and icons are unnecessarily small. I'm running at 225% scale in Windows settings. The windows can be manually resized, but this doesn't fix the icon problem and is simply a bit annoying. 

Here's what the windows look like on my machine:

[image]

[image]

[image]

t-paul, 2018-11-24:
That is something which should be handled by the underlying UI framework. Can you please try how the [development snapshots](http://www.openscad.org/downloads.html#snapshots) behave? Those should come with a newer version of that framework which is supposed to handle per monitor DPI scaling.

hrydgard, 2018-11-24:
The development snapshot behaves the same.

MichaelAtOz, 2018-11-25:
Which  Windows version?
I suspect W7? W7 doesn't handle high DPI properly, apparently W10 does, but I just live with small icons/fonts.

hrydgard, 2018-11-25:
Nah, this is Windows 10.

Except for the initial window sizes and small icons, it all seems to work fine, so this is a mild annoyance. Does smell like hardcoded window sizes in pixels somewhere though.

t-paul, 2018-11-25:
Nope, it's not hardcoded (well, the UI files have dimensions, but scaling DPI scaling should still work). It does work as advertised on Linux using the documented environment variable. If that is not set, I'm getting the same effect on my Dell XPS 13 @ 3840x2160.

Docs linked above say:
> Qt applications by default are Per-Monitor DPI Aware on Windows 8.1 or System-DPI Aware on older versions of Windows. As of Qt 5.4, the level can be specified by passing a parameter to the platform plugin (see Using qt.conf).

I guess we need to check if there's some unwanted config shipped somehow.

MichaelAtOz, 2018-11-26:
That page also  says 
_Applications mostly work with device independent pixels. **Notable exceptions are OpenGL** and code that works with raster graphics._
?

I can't recall what it did with W7 @ 150% DPI (to make fonts more readable), but I didn't like it. I'll try again.

ksuszka, 2023-12-01:
I installed a fresh installation of Windows 10 and OpenSCAD 2021.01, set system scalling to 225% and stumbled on the same issue. What solved it for me was setting QT_AUTO_SCREEN_SCALE_FACTOR environment variable to 1 before running OpenSCAD. This variable is described here: https://doc.qt.io/qt-5/highdpi.html#high-dpi-support-in-qt.

erjoalgo, 2024-11-10:
QT_AUTO_SCREEN_SCALE_FACTOR didn't seem to work for me, but QT_SCALE_FACTOR=3 did