openscad/openscad #2605: Changelog

sbridger, 2018-11-25:
There seems to be no (readily apparent) changelog listing changes and new features in each release.

- it should be linked from download page under heading "Development Snapshots"
-  It should be linked direct from the help menu or about dialog inside openscad
- its should be shown by the installer as is customary
- new features/changed language features should link direct to the specific wiki topic.
- language features in wiki should have the implemented-from version listed

t-paul, 2018-11-25:
There is a change log on the [releases](https://github.com/openscad/openscad/releases) page. The [snapshot cheatsheet](http://www.openscad.org/cheatsheet/snapshot.html) already links to updated documentation.

Other than that, there's nobody around who seems to have the time to maintain all that, so I'll close this issue as it's much too broad and unspecific. We may have specific issues that cover topics that can be closed, e.g. via some automation at build time or similar things.

closed 2018-11-25 by t-paul

sbridger, 2018-11-26:
Since that changelog is 3 1/2 years only, and has none of the last dozen releases, its not really doing the job is it?

The snapshot cheatsheat does not identify new features either.

So the issue: "no functional changelog" is totally unaddressed, so why close it?

t-paul, 2018-11-26:
The last release (2015.03) sadly is that old, I'm not sure what you mean by "last dozen releases". There is only a collection of changes for the [next release](https://github.com/openscad/openscad/blob/master/releases/2018.XX.md) which hopefully will happen sometime.

I tried to explain why it's closed. It's asking for too many things at once and at least the last one is covering a big continuous documentation effort that is not realistically going to happen. It would be great to have someone who maintains all that, but that was not the case for the last years.

Ok, so here's the long explanation

> Add a changelog file

This exists for releases only, I don't agree with the statement that it's customary to have change logs for the development process (other than the SCM history). It would be interesting to see if we can find a way to make the git history more usable for that though. The main thing here is it should not create extra effort.

> link to changelog from download page under heading "Development Snapshots"

Same reason, does not exist for dev snapshots. However there is some kind of staging area for the documentation which tries to track bigger features that are [WorkInProgress](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/WIP) but not fully integrated.

> linked direct from the help menu or about dialog inside openscad

That should be possible, but I don't see much reason to include that. What's the benefit? I don't really see that in a lot of software (actually none of the programs I've used today so far). Some commercial software might do that, but I never found that useful myself (I'm mostly using Linux, so maybe it's different on other platforms). For the dev snapshots that would likely include only the lists of the previous releases.
On the technical side, it's not ideal that the github change logs tend to be in Markdown format which probably would need conversion to be easily included in the application.

> add to the installer as is customary

That's possible, but is anyone really reading that at this point in time? Everyone I saw installing software just clicks next->next->next->finish so it would just add another annoying extra button click. Also this would be Windows-Installer only, the other platforms don't have an installer.

> new features/changed language features should have a link direct to the specific wiki topic.

Link where, from the change log? I think that's not matching the documentation well enough to be very useful, as the change log usually just mentions the general feature. The cheat sheet has a much better way of linking to additional documentation. 

> language features in wiki should have the implemented-from version listed

Yes, that would be nice to have, although it's getting complicated if it's not a feature but just changed a bit. The info is already there for most features that are not in the release (e.g. see angle parameter of [`rotate_extrude()`](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Rotate_Extrude).
Maintaining that is quite some effort though and it's also a continuous topic, so this makes the ticket impossible to close. This is more an item for a check-list.

MichaelPFrey, 2018-11-26:
> (other than the SCM history).

Actually, I think merged the pull requests to master would be pretty usable as a starting point.

t-paul, 2018-11-26:
Yep, that is how I updated [2018.XX.md](https://github.com/openscad/openscad/blob/master/releases/2018.XX.md) although the "squish & merge" makes it a bit harder.

sbridger, 2018-11-27:
Thanks for taking the time to reply. I will first nail my colours to the mast by saying: "An undocumented feature is _no feature_ " . If users do not know a feature exists, it doesn't exist - the developer has wasted his time: It will be unused, and in development snapshots, it will remain untested - voiding the ostensible purpose. 

So you understand that I "walk the talk", here is a [changelog of mine](https://realterm.sourceforge.io/change_log.txt). 
I am a single developer, and I can do this. Adding a few lines to the changelog as changes are made, is an absolutely negligible effort. (It does require some discipline)
As I believe the statement at the top, and I also agree that many people are not bothering to read the changelog during installs, I now make more effort, and copy this into the [News with a short summary](https://sourceforge.net/p/realterm/news/)

>  This is more an item for a check-list.
So where do I create such a checklist? Perhaps creating a "Documentation" [project](https://github.com/openscad/openscad/projects) might be appropriate?

>That's possible, but is anyone really reading that at this point in time? 
Obviously I am. And while new users don't bother looking at changes, upgraders definitely do want to see what is new. Often before bothering to install - so having it right above the "development snapshot download" is the most important place

MichaelPFrey, 2018-11-27:
We could help our-self by adding a todo list to the pull request template.

(At the same time: This can lead to funny bureaucracy - I have seen projects here on github were what is objectively a clean bug fix that addresses an obvious bug, but the PR does not get merged, because there is no test for it and there is no update to the changelog.)

> An undocumented feature is _no feature_ " . If users do not know a feature exists, it doesn't exist - the developer has wasted his time:

I am currently working on improving our error messages and warnings. The user does not need to know, that he know will get more context. Users will naturally find and use those. A "improved error messages" is enough in the changelog, even-tough those are multiple pull requests, dealing with various detail aspects.
Having to write a detailed changelog would drive me crazy.

Others are working on other big and important things, that really to the user are just a side note.

Fixing an obscure bug can be important - but still might not justify adding it to the change note, when the bug was just in development snapshots and effected only one user.

Fixing build errors also is something that does not need to be in the public changelog. There, an issue and PR on GitHub is fine.

This is the reason why I hesitate to add a PR-template. It would also make the first PR of newcommers  much more complicated. Currently, it is relatively easy and unbureaucratic - adding rules and process makes things harder.

nophead, 2018-11-27:
I think this issue is simply because there hasn't been a release for three years, so a lot of people are using development snapshots, rather than just developers. The actually releases have a changelog. They simple need to be more often. Perhaps concentrate on just one new feature and release it. Currently there are loads of changes being matured in parallel. When one of the todo items is completed, why not make a release instead of waiting for them all?

sbridger, 2018-11-27:
> I think this issue is simply because there hasn't been a release for three years, so a lot of people are using development snapshots, rather than just developers. The actually releases have a changelog. They simply need to be more often. 

Indeed. I see no advantage in (infrequent) snapshots , instead of a proper "development release". Having Stable Releases several years apart, and "Development Releases" in between is an excellent arrangement.