openscad/openscad #2607: Console output since the OpenSCAD 2018.11.20.ci571 version

STPR, 2018-11-26:
Hello,

I'm using the console as a summary of the parameters I choose in the customizer and it's usefull to distinguish them with colors.

But it appears the console behavior has changed since the OpenSCAD 2018.11.20.ci571 version.
Example with this simple code:
A = 200; // [1:1000]
B = 200; // [1:1000]
C = 200; // [1:1000]

// A (red):
echo("<b style='color:red'>");
echo(A=A);

// B (green):
echo("<b style='color:green'>");
echo(B=B);

// C (blue):
echo("<b style='color:blue'>");
echo(C=C);

echo("<b></b>");

[image]

Same code with the previous OpenSCAD 2018.10.13.ci145 version:
[image]

Also, I think it could be interesting to have an "Echo" window with just the echoes output, that would clear automatically after each rendering.

Thank you for OpenSCAD 👍

t-paul, 2018-11-26:
Yes, that changed via #2541 (as noted in #2606 too). Pinging @nophead - that's pretty much what I meant in the discussion - it's probably impossible to get the HTML stuff right.

I totally agree with your suggestion to have a separate "Output" window leaving the console with really just the log stuff generated by the application itself. That will break HTML output from scripts, but I don't see a good solution out of this as that is pretty much using an accidental implementation detail (Qt widget supporting limited HTML) which was then documented as feature.

nophead, 2018-11-26:
You can still have coloured output using the syntax described in the wiki manual. See https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#Echo_Statements

STPR, 2018-11-26:
Yes.
It works with simple code like this: echo("<b style='color:red'>", A=A);
It doesn't look as good as before but it works.

But when you have a more complex code with multiple scad/modules files with echoes which need to be of one color at one time and an other color later.

It doesn't work anymore.

t-paul, 2018-11-26:
Yes, and the problem is there is no good way to fix other problems and have the old HTML behavior, so I would close this issue - as the change is intentional - but we create a new issue for discussion how a useful Output-Window should behave (which likely includes the feature request for some basic formatting).

nophead, 2018-11-26:
I am not sure what is not working anymore. All the documented behaviour still works. Please can you give a more specific description of why it looks worse and what doesn't work.

The main difference is attributes only last for the echo. They don't leak out into the rest of the console.

STPR, 2018-11-26:
That's it.
The color don't leak to following echoes anymore, which was usefull.

But we need a better way to do that.
Something like a "font" or "format" command that would only work in a dedicated window for echoes and refreshed after each rendering.

nophead, 2018-11-26:
You could make a module that echos a string with a specified colour by using str() to wrap it in the appropriate HTML. Does that help?

In general having modal states does not fit well with a functional type language but echo is sort of procedural.

STPR, 2018-11-27:
It doesn't help since i'm already using str() inside my modules.
But i'm not sure to understand your solution correctly.

nophead, 2018-11-27:
Obviously I should have passed 1,2,3 in the last example.

If you don't want to pass extra parameters, perhaps if there are many level of code in between, then you can use $ variables to set the colour. E.g.
```
module colored_echo(text) echo(str("<b style='color:",$echo_color,"'>", text, "</b>"));

module Text(t){

if(t==1){ colored_echo(str("Text 1")); }
if(t==2){ colored_echo(str("Text 2")); }
if(t==3){ colored_echo(str("Text 3")); }

}

let($echo_color = "red") Text(1);
let($echo_color = "green") Text(2);
let($echo_color = "blue") Text(3);
```

STPR, 2018-11-27:
Interesting solution. Thank you.

But I think it complexify the code a little bit more.
And you have to write the same color for every single echo.

If you want  to keep the actual console behavior, please add a "console_refresh" command.

I still believe a dedicated echo window plus an echo_format command would be a better solution.

STPR, 2018-11-27:
I think the bold attribute is still leaking btw:

````
module colored_echo(text) echo(str("<b style='color:",$echo_color,"'>", text, "</b>"));

module Text(t){
if(t==1){ colored_echo(str("Text 1")); }
if(t==2){ colored_echo(str("Text 2")); }
}

let($echo_color = "red") Text(1);
Text(2);
````

Text 2 is black but still bold.

nophead, 2018-11-27:
It isn't leaking. The last call without setting the colour is still using the b element.  You could also test if echo_color is defined with is_undef() and remove the bold if no colour is specified.

A more flexible version could use a list with a colour plus formatting information. E.g, color, bold, italic, etc.

You don't need to set the colour any more times than you did before by using braces. E.g.

```
let($echo_color = "orange") {
Text(1);
let($echo_color = "blue") Text(2);
Text(3);
}
```

Will echo 1 and 3 in orange and 2 in blue.

You could even do: -
```
for(i = [1:3], $echo_color = ["red", "green", "blue"][i - 1]) Test(i);

Not sure what you mean by console_refresh. Do you mean a clear command? If you set the console size to the length of your output it achieves the same if the output is fixed size.

STPR, 2018-11-27:
Yes, it's not leaking sorry. I've removed my post.

I had forgotten the possibility to use {} with let and it works perfectly now 👍 .

It makes some lines longer when there are lot of variables in a single echo since the module colored_echo needs a string but it's acceptable...
````
colored_echo(str("A = ", A, ", B = ", B, ", C = ", C, ", D = ", D, ", E = ", E));
````
instead of just:
````
echo(A=A,B=B,C=C,D=D,E=E);
````
Yes a clear command.
I'm searching how to set the console size but it means some potentially important messages will not be visible anymore.
So i'm still wondering if it's a good idea to mix echoes and others message...

nophead, 2018-11-27:
By setting the size I just meant dragging the window divider.

What do you actually use the output console for? A better idea of your use case would help to know how it can be improved. I only using it for debugging and some design stats so it doesn't matter if they are mixed in with OpenSCAD's stats and I don't use HTML formatting.

STPR, 2018-11-27:
oh ok...I was searching for an option or a parameter...

Like I said in my first post, I actually use the console to summarize what I choose in the customizer.

If I choose a certain id (a number) with a slider in the customizer, some references, diameters must be displayed in the console. And It should be accurate.

But with the console and since it doesn't clear anymore, there's a risk that, if you scroll too much, you read the incorrect references and diameters from the previous renderings.
You constantly need to be sure you're at the bottom of the console and read the latests values.

This is why I think a dedicated echo window is needed with just the latests rendered values.

t-paul, 2018-11-27:
Why do you need that in the first place, maybe there's something that could be done in the customizer to remove the need for dumping stuff to the console? A value history maybe?

nophead, 2018-11-27:
The last messages should always be at the bottom, separated by a blank line
from the previous ones. I have never used the customiser, does it behave
differently?

t-paul, 2018-11-27:
It just triggers a normal preview after the values were changed, so it should be same as pressing F5 multiple times (but I have not actually checked yet). I'm just trying to understand why dump values that are visible in the UI into the console. Knowing the reason there might be alternatives for this specific case.

STPR, 2018-11-27:
@t-paul:
I think the customizer is already working great. Nothing to add there.
I don't see how and where you would display all those references and stuff inside the customizer.
And, for my case, it already takes the full window vertical size with the "Inline Details" option selected.

To be able to change the background colors of some parameters to distinguish them would be a plus.

I don't see the customizer as a good way to summarize.
I prefer to read lines of text.

@nophead:
It behave the same with the customizer.
But those 10 lines (+ a blank one) means I should make the console window bigger.

STPR, 2018-11-29:
To summarize some of my ideas, I've just made a fake screenshot with some annotations to show you how it would look like (with a simple example):
[image]
When you have a lot of variables, the colors in the customizer are really helpfull to target the correct slider or parameter.
Should I make a pull request ?

t-paul, 2018-11-29:
The features as such look like a useful extension, there's some things to consider.

* The Thingiverse style comment syntax for parameters should stay as it is, matching the behavior of their Customizer as much as possible. There are a couple of ideas for extended features that should combine into something like a "native markup" for use in the OpenSCAD Customizer GUI
* We probably want to separate the 2 features (customizer extension and formatted output)
* I wonder if a bit less coloring would be better (not all the background) as there's the risk of making things unusable if users are configuring a different color scheme.

STPR, 2018-11-30:
- Does something like this could work ?
````/*[A (in red)]*//*[1.0,0.5,0.5]*/````
It doesn't seem to annoy OpenScad but I don't know about Thingiverse.
I've tested here [http://customizer.makerbot.com/docs#example](url) but it seems limited...
- What do you mean by "separate" ?
-  You're right.
Perhaps a vertical line of few pixels at the left of every parameters would be enough.

STPR, 2018-12-21:
Hello,

While the nophead's solution works it does not permit the use of some special variables.
You need to set the objects's colors and the echoes's colors individually since echoes use an html syntax.

A new command could permit to set the same color, for an object and some echoes, using only one special variable.
Which would be easier to modify some colors and reduce the possibility of errors.

nophead, 2018-12-21:
You could make user module that calls color on its children and also sets a $ variable that sets the same colour in colored_echo().

MichaelPFrey, 2018-12-21:
> While the nophead's solution works, you need to set the objects's colors and the echoes's colors individually since echoes use an html syntax.

Where is the problem? color() supports HTML colors:
```
col="#FF00AA";

echo(str("<font color=",col,">",col,"</font>"));

color(col)
cube();
```
[image]

STPR, 2018-12-21:
@MichaelPFrey 
The hexadecimal notation is not explained in the documentation...
````
Note that the r, g, b, a values are limited to floating point values in the range [0,1] rather than the more traditional integers { 0 ... 255 }. However, nothing prevents you to using R, G, B values from {0 ... 255} with appropriate scaling: color([ R/255, G/255, B/255 ]) { ... }

Since version 2011.12, colors can also be defined by name (case insensitive). For example, to create a red sphere, you can write color("red") sphere(5);. Alpha is specified as an extra parameter for named colors: color("Blue",0.5) cube(5);
````
I don't know where you've found this but it works :)

@nophead 
That's an other solution.
I'll try that. Thank you.

closed 2026-03-13 by github-actions[bot]