openscad/openscad #2612: Problem compiling on Ubuntu 16.04.05

smartavionics, 2018-11-30:
Hi, I have been building openscad from the repo for some years with no problems but haven't been able to compile it recently because of the following errors, can you please advise as to what's wrong? Thanks.

````
g++ -c -m64 -pipe -DSTACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -O2 -I/usr/lib/x86_64-linux-gnu/libzip/include -I/usr/include/libxml2 -I/usr/include/freetype2 -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -DEIGEN_DONT_ALIGN -frounding-math -std=c++0x -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2018.11.30 -DOPENSCAD_SHORTVERSION=2018.11.30 -DOPENSCAD_YEAR=2018.0 -DOPENSCAD_MONTH=11.0 -DOPENSCAD_DAY=30.0 -DDEBUG -DSTACKSIZE=8388608 -DUSE_QOPENGLWIDGET -DENABLE_DBUS -DENABLE_JOYSTICK -DUSE_SCINTILLA_EDITOR -DENABLE_LIBZIP -DENABLE_OPENCSG -DENABLE_CGAL -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_DBUS_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ext/libtess2/Include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtPrintSupport -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtConcurrent -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtDBus -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Iobjects -isystem /usr/include/eigen3 -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o objects/Preferences.o src/Preferences.cc
src/Preferences.cc: In member function ‘void Preferences::init()’:
src/Preferences.cc:179:30: error: ‘prefsActionInput’ was not declared in this scope
  this->toolBar->removeAction(prefsActionInput);
                              ^
src/Preferences.cc:180:30: error: ‘prefsActionInputButton’ was not declared in this scope
  this->toolBar->removeAction(prefsActionInputButton);
                              ^
src/Preferences.cc: In member function ‘void Preferences::featuresCheckBoxToggled(bool)’:
src/Preferences.cc:276:31: error: ‘prefsActionInput’ was not declared in this scope
   this->toolBar->removeAction(prefsActionInput);
                               ^
src/Preferences.cc:277:31: error: ‘prefsActionInputButton’ was not declared in this scope
   this->toolBar->removeAction(prefsActionInputButton);
                               ^
src/Preferences.cc: In member function ‘void Preferences::setupFeaturesPage()’:
src/Preferences.cc:327:31: error: ‘prefsActionInput’ was not declared in this scope
   this->toolBar->removeAction(prefsActionInput);
                               ^
src/Preferences.cc:328:31: error: ‘prefsActionInputButton’ was not declared in this scope
   this->toolBar->removeAction(prefsActionInputButton);
                               ^
src/Preferences.cc: In static member function ‘static void Preferences::create(QStringList)’:
src/Preferences.cc:800:15: error: ‘class Preferences’ has no member named ‘AxisConfig’
     instance->AxisConfig->init();
               ^
Makefile:2949: recipe for target 'objects/Preferences.o' failed
make: *** [objects/Preferences.o] Error 1
g++ -c -m64 -pipe -DSTACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -O2 -I/usr/lib/x86_64-linux-gnu/libzip/include -I/usr/include/libxml2 -I/usr/include/freetype2 -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -DEIGEN_DONT_ALIGN -frounding-math -std=c++0x -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2018.11.30 -DOPENSCAD_SHORTVERSION=2018.11.30 -DOPENSCAD_YEAR=2018.0 -DOPENSCAD_MONTH=11.0 -DOPENSCAD_DAY=30.0 -DDEBUG -DSTACKSIZE=8388608 -DUSE_QOPENGLWIDGET -DENABLE_DBUS -DENABLE_JOYSTICK -DUSE_SCINTILLA_EDITOR -DENABLE_LIBZIP -DENABLE_OPENCSG -DENABLE_CGAL -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_DBUS_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ext/libtess2/Include -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtPrintSupport -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtConcurrent -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtDBus -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Iobjects -isystem /usr/include/eigen3 -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o objects/mainwin.o src/mainwin.cc
src/mainwin.cc: In constructor ‘MainWindow::MainWindow(const QString&)’:
src/mainwin.cc:203:38: error: ‘class Preferences’ has no member named ‘ButtonConfig’
         connect(Preferences::inst()->ButtonConfig, SIGNAL(inputMappingChanged()), InputDriverManager::instance(), SLOT(onInputMappingUpdated()), Qt::UniqueConnection);
                                      ^
src/mainwin.cc:204:38: error: ‘class Preferences’ has no member named ‘AxisConfig’
         connect(Preferences::inst()->AxisConfig, SIGNAL(inputMappingChanged()), InputDriverManager::instance(), SLOT(onInputMappingUpdated()), Qt::UniqueConnection);
                                      ^
src/mainwin.cc:205:38: error: ‘class Preferences’ has no member named ‘AxisConfig’
         connect(Preferences::inst()->AxisConfig, SIGNAL(inputCalibrationChanged()), InputDriverManager::instance(), SLOT(onInputCalibrationUpdated()), Qt::UniqueConnection);
                                      ^
src/mainwin.cc:206:38: error: ‘class Preferences’ has no member named ‘AxisConfig’
         connect(Preferences::inst()->AxisConfig, SIGNAL(inputGainChanged()), InputDriverManager::instance(), SLOT(onInputGainUpdated()), Qt::UniqueConnection);
                                      ^
src/mainwin.cc:373:16: error: ‘class MainWindow’ has no member named ‘fileActionExport3MF’
  connect(this->fileActionExport3MF, SIGNAL(triggered()), this, SLOT(actionExport3MF()));
                ^
src/mainwin.cc:387:8: error: ‘class MainWindow’ has no member named ‘fileActionExport3MF’
  this->fileActionExport3MF->setVisible(export3mfVisible);
        ^
src/mainwin.cc:502:12: error: ‘class Preferences’ has no member named ‘ButtonConfig’
  instance->ButtonConfig->init();
            ^
src/mainwin.cc:529:17: error: ‘fileActionExport3MF’ was not declared in this scope
  initActionIcon(fileActionExport3MF, ":/images/3MF.png", ":/images/3MF-white.png");
                 ^
src/mainwin.cc:609:17: error: ‘class QTextEdit’ has no member named ‘setMaximumBlockCount’
  this->console->setMaximumBlockCount(500);
                 ^
src/mainwin.cc: In member function ‘void MainWindow::consoleOutput(const QString&)’:
src/mainwin.cc:2901:18: error: ‘class QTextEdit’ has no member named ‘appendHtml’
   this->console->appendHtml("<span style=\"color: black; background-color: #ffffb0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
                  ^
src/mainwin.cc:2904:18: error: ‘class QTextEdit’ has no member named ‘appendHtml’
   this->console->appendHtml("<span style=\"color: black; background-color: #ffb0b0;\">" + QT_HTML_ESCAPE(QString(msg)) + "</span>");
                  ^
src/mainwin.cc:2909:19: error: ‘class QTextEdit’ has no member named ‘appendPlainText’
    this->console->appendPlainText(qmsg);
                   ^
[... 1 lines cut ...]
    this->console->appendHtml(qmsg);
                   ^
Makefile:3716: recipe for target 'objects/mainwin.o' failed
make: *** [objects/mainwin.o] Error 1
g++ -c -m64 -pipe -DSTACKSIZE=8388608 -fno-strict-aliasing -std=c++11 -O2 -I/usr/lib/x86_64-linux-gnu/libzip/include -I/usr/include/libxml2 -I/usr/include/freetype2 -I/usr/include/freetype2 -I/usr/include/harfbuzz -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -DEIGEN_DONT_ALIGN -frounding-math -std=c++0x -D_REENTRANT -Wall -W -Wno-unused-local-typedefs -fPIC -DOPENSCAD_VERSION=2018.11.30 -DOPENSCAD_SHORTVERSION=2018.11.30 -DOPENSCAD_YEAR=2018.0 -DOPENSCAD_MONTH=11.0 -DOPENSCAD_DAY=30.0 -DDEBUG -DSTACKSIZE=8388608 -DUSE_QOPENGLWIDGET -DENABLE_DBUS -DENABLE_JOYSTICK -DUSE_SCINTILLA_EDITOR -DENABLE_LIBZIP -DENABLE_OPENCSG -DENABLE_CGAL -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_DBUS_LIB -DQT_CORE_LIB - [... cut ...]
src/input/InputEventMapper.cc:155:34: error: ‘class Preferences’ has no member named ‘ButtonConfig’
             Preferences::inst()->ButtonConfig->updateButtonState(i,button_state[i]);
                                  ^
src/input/InputEventMapper.cc:159:29: error: ‘class Preferences’ has no member named ‘AxisConfig’
        Preferences::inst()->AxisConfig->AxesChanged(i,axisRawValue[i] + axisTrimValue[i]);
                             ^
Makefile:4161: recipe for target 'objects/InputEventMapper.o' failed
````

smartavionics, 2018-11-30:
````
scripts/check-dependencies.sh 
depname     minimum     found       OKness      
qt          4.4         5.5.1       OK          
qscintilla2 2.7         2.9.1       OK          
cgal        3.6         4.7         OK          
gmp         5.0         6.1.0       OK          
mpfr        3.0         3.1.4       OK          
boost       1.35        1.58        OK          
opencsg     1.3.2       1.4.        OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.2.92      OK          
glib2       2.0         2.48.2      OK          
fontconfig  2.10        2.11.       OK          
freetype2   2.4         2.6.1       OK          
harfbuzz    0.9.19      1.0.1       OK          
libzip      0.10.1      1.0.1       OK          
bison       2.4         3.0.4       OK          
flex        2.5.35      2.6.0       OK          
make        3           4.1         OK          
Warning: you have pkgconfig under /usr/local/lib
````

t-paul, 2018-11-30:
From the error messages, it sounds like the `moc` generated files are somehow not correct. Maybe try again with a fresh clean source tree and also record the full build log.

smartavionics, 2018-11-30:
Hi, thanks for the response. I had already tried "make clean" and re-running qmake but it made no difference. Removing objects/moc_* seemed to do the trick. Perhaps the "make clean" rule should remove those files?

Anyway, I know now what to do so closing this issue...

closed 2018-11-30 by smartavionics