openscad/openscad #2613: Colored tabs inside the customizer to better distinguish the parameters

STPR, 2018-11-30:
Hello,
I think it could be interesting to have some colored tabs inside the customizer to better distinguish the parameters.
So, I've made a fake screenshot with some annotations to show you how it would look like (with a simple example):

[image]

For a red tab, the syntax could look like that:
````/*[A (in red)]*//*[1.0,0.0,0.0]*/````

It doesn't seem to annoy OpenSCAD but I don't know about Thingiverse.
I've tested here http://customizer.makerbot.com/docs#example but it seems limited because there's no tabs at all...

t-paul, 2018-12-01:
As mentioned in the discussion earlier (in #2607) this needs to be based on a more general customizer markup. The comment syntax as used by [Thingiverse](http://customizer.makerbot.com/docs) will not be extended. See related [wiki page](https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases) for some initial ideas.

STPR, 2018-12-01:
Ok.
Why not just add some new reserved special variables ?
````
/*[Hidden]*/
// No color for the first tab
$ctab_002 = [1.0, 0.0, 0.0];
$ctab_003 = [0.0, 1.0, 0.0];
$ctab_004 = [0.0, 0.0, 1.0];
````

t-paul, 2018-12-01:
What if we want to extend to other properties? More reserved variables with pattern naming makes things just complicated and hard to explain to users. A clearly defined way of adding meta information is much more useful and in the end even easier to implement and maintain.

STPR, 2018-12-21:
Just to add some thoughts on why some reserved special variables could be a useful solution:

It would allow to reuse the same colors later in the code for the objects.
You set the differents colors only one time at the start of the program which reduce errors.

What do you mean by others properties ?
Some colors would be enough to distinguish the different tabs.

MichaelPFrey, 2018-12-21:
> What do you mean by others properties ?
> Some colors would be enough to distinguish the different tabs.

Some examples of what people did or wanted to do:
- Some want translation support: #2434
- It used to be possible to use HTML in the description: #2193 (which is fine of OpenSCAD, but would be horrible for the thingiverse customizer)
- I have code that allows the tabs tobe conditionally shown: #2359 all it needs is a good way to implement it in syntax

Using #2359 as I base, I think I could get something like

```
/*[A (in red)]*/*color(colorA)**/
...
/*[B (in blue)]*/*color(colorB)**/
...
colorA=[1,0,0];
colorB="blue";
```

to work (colorA and colorB being just example variable names).

The issue is: **This is a syntax hack, on top of a syntax hack we do not control/own.**
We could develop our custom syntax - but someone would have to do it.
The idea is, that the our syntax can build on top of the current customizer - a lot of backend stuff would be the same - but that would still be a lot of work that does not yield results until much much latter.

t-paul is hinting at the first step:
> A clearly defined way of adding meta information is much more useful and in the end even easier to implement and maintain.

That sounds very complicated, technical and bureaucratic - because it is. I mean: Some software architects can do such things within weeks, it is not rocket science, but it is not a beginners task either. This syntax will shape OpenSCAD for ever.

Each time I am working on the customizer, I am more and more impressed, that a student has managed to setup the whole system as a Google Summer of Code project. There are a lot of things going on.

t-paul, 2018-12-21:
It's not really complicated. Most of that was even in the prototype for the Customizer already - just have a look at the two short screencasts linked in https://github.com/openscad/openscad/issues/722#issuecomment-69521189 to show 2 different use cases (more in the [wiki](https://github.com/openscad/openscad/wiki/Meta-Data-Use-Cases)). First one is what evolved into the Customizer (but switching to the comment syntax), the second one shows how it would be possible to present a list of objects from libraries for drag&drop.