openscad/openscad #2623: Appimage: openscad: symbol lookup error: undefined symbol.

AndKe, 2018-12-06:
./OpenSCAD-Nightly-0-Build2665.1.glibc2.14-x86_64.AppImage 

openscad: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers

Unable to run older appimage as well. (OpenSCAD-Nightly-0-Build2252.1.glibc2.14-x86_64.AppImage)

Snap works - that is; snap does not see and understand "home" directory, always showing root, it's less convenient.

t-paul, 2018-12-06:
Yes, that seems to affect newer distros (e.g. Debian/Testing). This depends on an update on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) (see https://github.com/AppImage/AppImages/issues/348).

Due to https://github.com/openSUSE/obs-service-tar_scm/issues/264 we could not build new AppImages for a couple of days, but this problem was fixed just now.

AndKe, 2018-12-06:
This is Ubuntu 18.10  
So, I should just try latest appimage in a few days ?

t-paul, 2018-12-06:
I just tried the new build and this still has the issue.

We have no control over OBS deployments, so I can't say much more than just retrying now and then. I'm thinking about creating builds independent from OBS, but that requires quite some effort so that will not happen very soon.

AndKe, 2018-12-07:
sadly, I can't use OpenSCAD at work,
-  have this very issue with appimage, 
- and https://github.com/openscad/openscad/issues/2378 with snap.
Seems to work fine @home (maybe due to Nvidia graphics - same distro both places)

t-paul, 2018-12-07:
The normal distribution specific packages are not useable?

t-paul, 2018-12-07:
Oh? I have heard about that resizing issue only on Windows so far (see  #2536). If that affects also Ubuntu 18.10 that's annoying. I do wonder why I'm not seeing that in Debian/Testing then.

t-paul, 2019-01-10:
Fixed with new AppImage build.

closed 2019-01-10 by t-paul

yunfan, 2019-01-25:
i met this too, and you said it has been fixed with new appimage build, but what i see in release is older than this issues's close

t-paul, 2019-01-25:
The 2015.03 release AppImage will not be updated. See [development snapshots](http://www.openscad.org/downloads.html#snapshots) section of the download page.