openscad/openscad #2625: Release, please? :)

ulab, 2018-12-07:
Can us "regular" users have a new official release, please? 

The latest one is from 2015-03 and a lot of things I have stumbled over as a new user seem to have been fixed already. Things like "View all" not zooming correctly or SVG exports not having units for example.

While developer snapshots might be a solution, it's just something us regular folks aren't really keen of. For example I just used brew to install on my mac and wouldn't have expected that there are snapshots that fix a lot of stuff without being released into the open.

I understand this has been asked about a year ago in #2225 already and I understand developer resources are slim, but an official release as old as it is will make the project look abandoned at first glance and might deter new users and possible interested developers too.

t-paul, 2018-12-07:
Please don't use the issue tracker as discussion forum. There's a [forum/mailing list](http://www.openscad.org/community.html#forum) for that (which not surprisingly has [some discussion](http://forum.openscad.org/Is-there-an-official-release-scheduled-and-when-tp24573.html) on that topic already).

Progress regarding a release can be observed in the [Project Tracker](https://github.com/openscad/openscad/projects/1).

closed 2018-12-07 by t-paul

ulab, 2018-12-07:
Yes, like I did and do with other open source projects too, but I don't know enough about the project yet since I have just started a few days ago. And I am not a coder, but I can test, help document and report issues I stumble over. And when I stumble over things and want to report issues that have been fixed two years ago, it feels like this could've been avoided.

So the one thing I started with was making a suggestion which would've avoided me looking into issues that are already fixed and might help increase the user base. I would not have done that if I wasn't interested in the project going forward.

What issues are open / blocking a release? Are the developer snapshots that unstable?

PS: Thanks, @t-paul for those links, those help.

t-paul, 2018-12-07:
The dev snapshots are usually pretty stable and usable, but of cause there's always a risk that issues could occur that are not caught by the test suite (e.g. UI glitches / #2501).

The blocking issues are listed in the [Project Tracker](https://github.com/openscad/openscad/projects/1). Basically it's about wrapping up the big changes in a releasable state. It will not be possible to cover every possible improvement or the next release would be in 10 years. Not an easy decision though :-).