openscad/openscad #2629: Language support for popular code editors via Language Server Protocol

akaleeroy, 2018-12-10:
People use their favorite editor because they've customized it to suit their needs and learned the ropes.
But OpenSCAD language support for various text editors is spotty and the landscape is [woefully fragmented](https://github.com/thenewhobbyist/OpenSCAD-Sublime-2/network/members "Forks · thenewhobbyist/OpenSCAD-Sublime-2") at present. For example, [OpenSCAD support for Sublime Text in Package Control](https://packagecontrol.io/packages/OpenScad) is incomplete and out of date, yet is relied on for ports to other editors.

This is what the **Language Server Protocol** was built to fix: https://langserver.org/

This issue is [meant to centralize discussion](https://xkcd.com/927/ "🙄") about OpenSCAD language support... I wanted to contribute a few small fixes (remove deprecated `assign()`, correct `multimatrix` typo etc.) and found it hard to even figure out where.

More discussion in #2299 and #2409 and [OpenSCAD - OpenSCAD language plugin for the IntelliJ platform](http://forum.openscad.org/OpenSCAD-language-plugin-for-the-IntelliJ-platform-td24643.html)

t-paul, 2018-12-10:
Yep, I'm totally in support for that (which is why I was asking in that linked forum thread hoping for work going into this direction :smile:).

However this will not really cover #2299 or #2409 as those were going more into the direction of making OpenSCAD itself support other stuff (kind of being a client for other services). That seems too far out of reach to even leave the feature request sitting here right now.

So to clarify, this ticket would cover the LSP server implementation which enables other IDEs/Editors to support the OpenSCAD language via calls to a running OpenSCAD instance.

akaleeroy, 2018-12-10:
> So to clarify, this ticket would cover the LSP server implementation which enables other IDEs/Editors to support the OpenSCAD language via calls to a running OpenSCAD instance.

Yep, that was the intention.
I mentioned those other issues and forum post just as a way to consolidate the conversation around this topic.

Lenbok, 2020-05-18:
LSP support would be fantastic. It seems like some of the back-end bits are starting to fall in place, with #3283. I reckon an LSP server for OpenSCAD would be a great GSOC project!

waynevanson, 2024-03-31:
Hey all, I've been looking into [creating a formatter for OpenSCAD using topiary](https://github.com/waynevanson/topiary/tree/openscad) and found out that creating an LSP may be possible using tree-sitter.

- tree-sitter grammar sourced from https://github.com/bollian/tree-sitter-openscad (seems complete)
- tree-sitter grammar to LSP from https://github.com/neomutt/lsp-tree-sitter (lots of examples, haven't had a look yet so not sure if it's fit for purpose).

Lenbok, 2024-03-31:
@waynevanson https://github.com/Leathong/openscad-LSP exists now.

t-paul, 2024-03-31:
It would be nice to collect those external pointers somewhere, e.g. in some appendix in the documentation on Wikibooks. This issue here is for native support in the OpenSCAD application itself. Prototype see https://github.com/openscad/openscad/pull/3635 linked above.