openscad/openscad #2631: if else in function

Parkinbot, 2018-12-12:
Just updated from 2018.04.06 to 2018.11.20.ci571 but still get a compile error for  
```
function test(a=0) = 
  [for(i=[-3:3]) 
    echo(i)
    if (a>0) a else -a]; 
```

I would also expect to be able to compile

```
function test(a=0) = 
   if (a>0) a else -a; 
```

nophead, 2018-12-12:
For the first one moving the echo after the if seems to fix it.

For the second one you would need the ternary operator. The list comprehension extensions only work in lists.

Parkinbot, 2018-12-12:
Thanks for the first one. 
The second one is fixed by
`function test(a=0) = [if (a>0) a else -a][0]; `

nophead, 2018-12-12:
The first one is a bit odd beacuse when it is after the if it needs to be in both branches. Possibly a bug that echo can't go before if but the echo function is wierd in that it can have an expression after it.

For the second one:
```
function test(a=0) = a > 0 ? a : -a; 
```
Is simpler. It also works in the first one, with the echo:
```
function test(a=0) = 
  [for(i=[-3:3]) echo(i)
    a > 0 ? a : -a]; 
```

I am not sure why if-else was added to list comprehensions. Can it do anything that the ternary oparator doesn't allow?

t-paul, 2018-12-12:
It's not a bug as `echo()` is not a list comprehension expression (as listed in [Basic Syntax](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/List_Comprehensions)), it's just a function call that shares some syntax with list comprehensions. Due to the shared syntax I suspect it would be trivial to add though.

Parkinbot, 2018-12-12:
Even more wierd: the echo in the following code will not only respond the true branch of the if clause, but to both. So how do I echo values only for the true branch? 

function test(a=0) = 
  [for(i=[-3:3] ) 
    if(a > 0) echo(i) 
    a else -a 
      ];

t-paul, 2018-12-12:
It does not, you are testing `a` not `i`in the `if`.

Parkinbot, 2018-12-12:
Oh sorry. How stupid.

closed 2018-12-12 by Parkinbot

nophead, 2018-12-12:
>yes you can omit the else part.

You can also do it this way, although it is a bit contrived and each came after if I think.
```
l = [for(i = [0:10]) each i % 2 ? [i] : []];
echo(l);
```
```
ECHO: [1, 3, 5, 7, 9]
```