openscad/openscad #2632: echo in rekursive function crashes OpenSCAD

Parkinbot, 2018-12-13:
The following endless recursion crashes at least 2018.11.20.ci571 because of echo I assume. 

```
a = rek(10); 
function rek(a) = echo(a) rek(a); 
```

kintel, 2018-12-15:
On what platform? 2018.12.05 on Mac gives me:
```
ERROR: Recursion detected calling function 'rek' in file issue2632.scad, line 2
```

MichaelAtOz, 2018-12-15:
Crashes for me on W7.

MichaelAtOz, 2018-12-15:
OpenSCAD Version: 2018.11.04.ci339 (git 29742ad7) latest on hand.

MichaelPFrey, 2018-12-15:
crashes for me on Ubuntu 18.10 using the nightly builds that t-pauls provides.
```
OpenSCAD Version: 2018.12.08
System information: Linux 4.18.0-12-generic #13-Ubuntu SMP Wed Nov 14 15:17:05 UTC 2018 x86_64 Ubuntu Core 16 4 CPUs 11.65 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
```
(it also crashes with other builds on my machine, but t-pauls nightly builds are somewhat more offical then my messy builds)

Parkinbot, 2018-12-15:
crashes for me on Windows 8.1 and 10.

MichaelPFrey, 2018-12-24:
```
a = rek(10); 
function rek(a) = echo(a) rek(a); 
```
this is not just a recursive function, this is a tail recursive function.
For tail recursion, we only have a counter limiting the recursion depth.
This makes sense, as a true tail recursion does not use extra stack space for each level of the recursion.
How ever: Echo does use stack space.

This also explains why kintel under MacOSX did not see it: We use a different compiler for OSX and operating system. Both are valid variables that influence both the stack size and the stack usage.

nophead, 2018-12-24:
Presumably that means it isn't detected as tail recursion due to the presence of echo because echo in a loop should not use more stack than a single call.

MichaelPFrey, 2018-12-24:
> Presumably that means it isn't detected as tail recursion due to the presence of echo

It is detected as a tail recursion. Look at #2647 which works as a fix on ubuntu 18.10.
I would have written a test case, but it leaks so little stack per call, that it takes some time until it either detects an error or crashes.

> because echo in a loop should not use more stack than a single call.

You are hinting at the fact, that the [echo() function](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Echo_Function) (not be confused with the [echo() statement](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#Echo_Statements)) might be buggy. But at some point: The string to be outputted by echo has to be stored and passed on "somewhere" and "somehow". Somewhere is either heap or stack. "Somehow" is almost certainly via method calls, using stack. Sooner or latter, this lets to running out of memory. Blocking the echo function in tail recursive function would work has a hot fix - but the echo function is specifically there for debugging such cases.

We also could have or latter add other functions that unintentionally and slowly eat up memory.

MichaelPFrey, 2018-12-25:
I am confused. Yesterday when I experimented it crashed reliable. Now it does not crash.
My assumption that it has do with tail recursion is wrong.
But now I start to find separate issues.

thehans, 2020-06-04:
I am unable to reproduce a crash in Ubuntu 18.04 with latest snapshot.  
I believe that @MichaelPFrey's  stack changes have solved this some time ago.

Can the other users who experienced this(particularly in Windows?) verify if its fixed for them?

nophead, 2020-06-04:
Fixed on Win7 64.
```
ERROR: Recursion detected calling function 'rek' in file RepRaptor, line 2 
TRACE: called by 'rek', in file RepRaptor, line 1. 
```

However it does show a new bug. The file is Untitled.scad. RepRaptor is a directory that it the default it offers for Save but it isn't where I save my other scad files, it is two level above most of them.

thehans, 2020-06-04:
OK I'm gonna call this fixed then.  

For the directory vs Untitled issue, see #2878

MichaelAtOz, 2020-06-05:
On Windows 7/64, with 2020.05.06, this just keeps logging `ECHO: 10` using 100% of a core.
Then tried Debian nightly 2020.06.04, same but less CPU.

thehans, 2020-06-06:
@MichaelAtOz 
Well, on Ubuntu 18.04 the logging was also pretty slow for me, at maybe 30 lines per second.
So it would take quite a while to hit get to the "Recursion detected", which happens at 1million lines!  

Fortunately I found that hiding the console makes it go MUCH faster since it doesn't really have to redraw the GUI.  It still takes about 2 minutes on my desktop, but I can watch CPU usage in task manager to tell when its done, then un-hide the console and see the "recursion detected" error, but most importantly no crashing.

I also made a slightly modified version of the original example, so progress can be tracked:
```
a = rek(0); 
function rek(a) = echo(a) rek(a+1); 
```

If you can verify that there's no crash at least, then please re-close.

I think the console logging speed could definitely be improved, possibly by buffering lines and throttling widget updates to no more than X times per second.  So each update could push out multiple lines at once for situations like this.  But I would open that as an separate issue from this, which is specifically reporting a crash.

MichaelAtOz, 2020-06-06:
```
ECHO: 999999
ECHO: 1e+6
ERROR: Recursion detected calling function 'rek' in file mebd, line 1 
TRACE: called by 'rek', in file mebd, line 2. 
Compiling design (CSG Products generation)...
```

closed 2020-06-06 by MichaelAtOz

rcolyer, 2020-06-06:
For future reference I'm going to document that the difficulty reproducing this might be due to a thread race condition happening on calls to MainWindow::consoleOutput, which currently calls this->processEvents(); at the end.  This was initially added out of a performance motive, but it's a suspicious place to have that call, because if a very rapid stream of console output events (like this example) is sent faster than it can process them, this could result in infinite recursion into processEvents followed by a stack overflow.  If somebody at some point ends up able to reproduce this in the current code as a crash on their platform or testing conditions, that would be a line to try changing for a fix.

MichaelAtOz, 2020-06-07:
Do you want to know about a crash, when you close the OpenSCAD window while it is doing all that echo()ing?

t-paul, 2020-06-07:
That might be related to #1251.