openscad/openscad #2634: IDF (Intermediate Data Format) import

Knochi, 2018-12-14:
Hi,

I'm an electronics guy with little experience in C++. I would like to be able to import IDF Files. These is a common interchange format form eCAD to mCAD. 
IDF export has to files:
- .emn contains outline of PCB and positions of holes and parts. 
- .emp contains outline and height of parts

I did an import script for PCB in Blender several years ago and remember it wasn't that hard. Just a bunch of polygons, holes and extrusions.

Wikipedia Article has a few nice links https://en.wikipedia.org/wiki/Intermediate_Data_Format

MichaelPFrey, 2018-12-14:
I am a PCB design engineer my self and I know the IDF Fileformat.

One argument for IDF support is KiCAD.

But I do not think that adding the ability to import IDF is the right approach.
Writing a convert in any script language from IDF to CSG (a subset of the Openscad Language) would be trivial.
Using scripts allows the user to easily hack the script. For example, when you seriously use IDF as exchange format, companies start to map more detailed models that replace the oversimplified the models from the emp files.

Something that I had to deal with professionally, was crap IDF Files from a vendor. It was "easy" to fix, but it involved scripting to look at the various board outlines and manually editing both emn and emp file and it showed me that the designer had no idea how to use his own PCB design software. That is not uncommon. Using an import script instead of an import allows for a very flexible adaptation.

Fun fact: [that is my contribution the IDF wikipedia article](https://en.wikipedia.org/w/index.php?title=Intermediate_Data_Format&type=revision&diff=801038001&oldid=740882075)

t-paul, 2018-12-14:
Yes, it would be much better adding support for STEP, but that is a *huge* project and STEP is a massive beast even with existing libraries. I think I looked at this before (maybe due to some discussion on IRC, I don't remember the details) and the conclusion was similar.

That said, with the better solution (probably) way out and not much alternative, maybe it's worth a shot. There seems to be a pretty straight forward usecase where this could help. Creating 3D-printed enclosures for PCBs. If this can give a simple way to get the PCB outline transfered into OpenSCAD as placeholder it might be useful for a number of people even with the attached limitations.

kintel, 2018-12-17:
I didn't look at your code for arcs, but we have DXF arc code in OpenSCAD that you could look at:
https://github.com/openscad/openscad/blob/master/src/dxfdata.cc#L208

Knochi, 2018-12-17:
Thanks! That helped a lot... implemented the facets function as a openscad function and calculated the fragments..

```
polygon(arcGen(P3,P4,-90));

function arcGen (P1,P2, angle, iter=0, result=[])=
  let (
    dist=norm(P1-P2),
    arcRadius= (P1.x<=P2.x) ? dist/(2*sin(-angle/2)) : dist/(2*sin(+angle/2)), //radius of the Arc
    fragments=getFragmentsFromR(arcRadius), //number of fragmets to draw the arc with resulting from fn,
    alpha1= (P1.x<=P2.x) ? (180 + angle)/2 : (180 - angle)/2,
    alpha2=asin((P1.y-P2.y)/dist),
    beta=alpha2+alpha1-90,   
    Mx=(P1.x<=P2.x) ? P1.x - sin(beta)*arcRadius : P1.x + sin(beta)*arcRadius,
    My=P1.y - cos(beta)*arcRadius,
    M=[Mx,My],
    //result = (result==undef) ? [M] : result, //initialise with M
    angOffset=(P1.x<=P2.x) ? 180-alpha1 - alpha2 : alpha1 +alpha2,
  
    x = arcRadius * cos(angOffset+angle/fragments*iter),
    y = arcRadius * sin(angOffset+angle/fragments*iter),
    
    steps = ceil(fragments * abs(angle)/360), // abs(($fn/360)*angle), //using ceiling function helps sometimes but not everytime
    stepSize = fragments/steps
  )
  (iter<=fragments) ? arcGen(P1,P2,angle,iter+stepSize, concat(result,[M+[x,y]])):result;
    
function getFragmentsFromR(r)= 
  ($fn>0) ? $fn : ceil(max(min(360/$fa,r*2*PI/$fs),5));

```

t-paul, 2018-12-18:
Hmm, something like that:

[image]

I'm toying with Qi parsers, so that's a nice test case which is also useful at the same time :smile:. I guess we are talking IDF3 which seems to be implemented in KiCAD. IDF4 looks totally different, so that would need a completely different import.
The board used as example is the [ICEBreaker](https://github.com/icebreaker-fpga/icebreaker) v1.0c by @esden.

nophead, 2018-12-18:
> useful at the same time 😄. I guess we are talking IDF3 which seems to be

Nice.

Is there a bug where the holes that make the perorations break out?

t-paul, 2018-12-18:
Well, I'd call it a not implemented feature :wink:. Yes, putting everyting into a single polygon structure causes glitches, not sure how to handle some of those edge cases yet. Maybe generating a CSG tree with the holes separate. We discussed something like that anyway to have some means of having a stable import (by producing temporary CSG files that could be cached).

t-paul, 2018-12-19:
So that is what I came up with for calculating the arc case based on my limited math knowledge :smile:. It seems to work for both postive and negative angles including 180 degrees, but there's probably some room for improvment...
https://github.com/openscad/openscad/blob/967b0d73045fea896dcaeb93eeca7e85716d081c/src/import_idf3.cc#L171-L186

nophead, 2018-12-19:
Maybe passing it through clipper would sort out the holes intersecting the outline.

Knochi, 2018-12-20:
So it's here! My first version of a javascript parser. For now it comes without a website, just put your IDF files in the data directory, tweak the code and you get a wonderful .scad file rendering your board.

https://github.com/Knochi/IDF2OpenSCAD

let me know what you think.

Knochi, 2019-01-10:
Uploaded the sketch to a free webhoster service. You can play with it!
http://knochi.mygoodpage.org/

Knochi, 2019-12-23:
Agree on that STEP import/export is more usefull

closed 2019-12-23 by Knochi