openscad/openscad #2640: Crash with --debug & output redirection

MichaelAtOz, 2018-12-22:
`"C:\Program Files\openscad-2018.12.08.ci706\openscad" --debug=all cmd-line.scad -o cmd-line.echo > cmd-line.txt`

Regardless of .scad code, `-debug=all` without redirection, or redirection without `debug=all` works, but both crashes.

MichaelAtOz, 2018-12-22:
Windows 7, multiple OpenSCAD versions, 2018.12.08 is the latest I have at hand.

MichaelPFrey, 2018-12-22:
Ubuntu 18.10:
```
./openscad --debug=all cmd-line.scad -o cmd-line.echo > cmd-line.txt
Debug on. --debug=all
Camera: Camera()
openscad: BuiltinContext:
Context: 0x7ffdc26f38b0 (0)  document path:   vars:    PI = 0x55a868229ce0    $vpr = 0x55a86820e610    $fs = 0x55a86820d860    $fn = 0x55a86820da90    $vpd = 0x55a86820e700    $fa = 0x55a86820d7f0    $t = 0x55a86820dec0    $preview = 0x55a868212b20    $vpt = 0x55a86820e520
```

MichaelPFrey, 2018-12-22:
https://github.com/openscad/openscad/blob/f2848902e8264f2fd068694741de6aaaec33f452/src/openscad.cc#L352-L354
https://github.com/openscad/openscad/blob/f2848902e8264f2fd068694741de6aaaec33f452/src/context.cc#L223-L256
`v.second` has to become `v.second->toEchoString();` which is better:
 ```
./openscad --debug=all cmd-line.scad -o cmd-line.echo > cmd-line.txt
Debug on. --debug=all
Camera: Camera()
openscad: BuiltinContext:
Context: 0x7ffdf53055c0 (0)  document path:   vars:    PI = 3.14159    $vpr = [0, 0, 0]    $fs = 2    $fn = 0    $vpd = 500    $fa = 12    $t = 0    $preview = false    $vpt = [0, 0, 0]
```
now we add a few new lines:
```
./openscad --debug=all cmd-line.scad -o cmd-line.echo > cmd-line.txt
Debug on. --debug=all
Camera: Camera()
openscad: BuiltinContext:
Context: 0x7ffc9a68bad0 (0)
  document path: 
  vars:    PI = 3.14159
    $vpr = [0, 0, 0]
    $fs = 2
    $fn = 0
    $vpd = 500
    $fa = 12
    $t = 0
    $preview = false
    $vpt = [0, 0, 0]
```
now I wonder how
```
		s << boost::format("Context: %p (%p)\n") % this % this->parent;
```
is supposed to work.

I'll open a Pull request. But please look for other possible causes - I might be working on a separate issue, as I am not certain if I even show a crash.

MichaelAtOz, 2018-12-22:
With the crash this is the output
```
Debug on. --debug=all
Camera: Camera()
```

MichaelAtOz, 2018-12-22:
Above is the redirected output (.txt) file.

This is the .scad I use
```
echo(arg0=arg0);
echo(version());
echo("Hello world! ");
//linear_extrude(height=1) text("hello world!");
```
This is the .echo
```
WARNING: Ignoring unknown variable 'arg0', in file cmd-line.scad, line 1.
ECHO: arg0 = undef
ECHO: [2018, 12, 8]
ECHO: "Hello world! "
```

MichaelPFrey, 2018-12-22:
looking at your output: The crash looks to be windows specific.

MichaelAtOz, 2026-03-15:
For the record, this crash doesn't happen with 2026.02.19.