openscad/openscad #2644: module abstraction.

skitov, 2018-12-22:
I don't know how hard would be the implementation, but it would be very useful to have some form of module abstraction, may be something similar to C++ templates. Some hypothetical code example:
```
module <abstract_module>xxx()
{
    do_something() abstract_module(param_list);
}
module implementation1(param_list){}
module implementation2(param_list){}
<implementation1>xxx();
<implementation2>xxx();
```

t-paul, 2018-12-22:
I suspect most cases can be handled using `children()`.

Otherwise please add some specific use cases, that would help trying to find possible solutions.

skitov, 2018-12-23:
Ok, some more concrete examples.
Well known problem with adding a shape, preserving hole in it:
```
module <abstract_shape_with_hole>add_shape_preserving_hole()
{
    difference()
    {
        union() { children(); abstract_shape_with_hole(hole=false); }
        abstract_shape_with_hole(hole=true);
    }
}

<one_shape_with hole>add_shape_preserving_hole() 
<another_shape_with_hole>add_shape_preserving_hole()
    cube(100);
```
Grid of parametric shapes, in this example we want to make a grid of parametric shapes into a grid:
```
module <abstract_parametric_shape>parametric_grid()
{
    for(ix=[0:4], iy=[0:4])
    {
        x=xpos(ix);
        y=ypos(iy);
        p=param(ix,iy);
        translate([x,y,0]) abstract_parametric_shape(p);
    }
}
```

t-paul, 2018-12-23:
Both cases already have solutions.
```
module diff() {
    difference() {
        let($positive = true) children();
        let($positive = false) children();
    }
}

module array(o) {
    for (x = [-1, 1], y = [-1, 1]) {
        translate(o * [x, y]) children();
    }
}

module a() {
    if ($positive) {
        cube(12, center = true);
    } else {
        cylinder(r = 4, h = 30, center = true);
    }
}

module b() {
    if ($positive) {
        sphere(r = 10, center = true);
    } else {
        cube([5, 5, 40], center = true);
    }
}
diff() {a(); array(12) b(); };
```
I do agree that this really could use improvement as the use of special variables is a bit obscure. I can't see yet how the suggested feature is much different. Note that there is a [proposal for object inheritance](https://github.com/doug-moen/openscad2/blob/master/rfc/Inheritance.md) which would better fit into the existing OpenSCAD syntax.

skitov, 2018-12-23:
I'm sorry, but I do not see solutions for both of provided problems. Array() places non parametric objects, exact copies of all the nodes of the array. Let's assume that you need to attach a() to an arbitrary shape  so, that it preserves cylindric hole in the cube, how your diff() module will help?

skitov, 2018-12-23:
I did the same in my comment, so for every module you are going to add holed module to you need to care about $positive handling, that is basically is some amount of copy-paste.

t-paul, 2018-12-23:
How is that different in your proposal? You have to state *somewhere* what the positive and what the negative part is. There is no copy-paste though, every module states it's own part.

skitov, 2018-12-23:
If you want to attach nuttrap() to let's say cube([20,20,2], center=true) if you do not add $positive handling to the cube, it will make an additional hole in nuttrap() instead of what is wanted. And you will need to add $positive handling for every shape you add nuttrap to, unnecessary copying and pasting of "if($positive)". In my proposal you need to handle about positive for those shapes, where it is needed only (for nuttrap).
```
module <holed_module>add_holed_module()
{
	difference()
	{
		union()
		{
			children();
			let($positive=true) holed_module();
		}
		let($positive=false) holed_module();
	}
}
```
<nuttrap>add_holed_module() shole expand to
```
module nuttrap_add_holed_module()
{
	difference()
	{
		union()
		{
			children();
			let($positive=true) nuttrap();
		}
		let($positive=false) nuttrap();
	}
}
```

skitov, 2018-12-23:
The bad thing in your diff()/combine() example, that children() are not homogeneous, and this approach with assigning different roles to different children should be avoided, in my opinion.

skitov, 2018-12-23:
After some thinking, I agree that grid problem is perfectly solved with special variables, but hole preserving solution has some degree of ugliness.

t-paul, 2018-12-23:
Yes, I agree. Although I don't see how this complicated and a bit alien concept (e.g. OpenSCAD does not have user defined types) is the way to go. There are some benefits but also a number of downsides. Maybe combining the idea with the Object proposal linked above could improve things?

t-paul, 2019-11-14:
More specific discussion in #3088, so I'm closing this one as it's still too high level to result in any actions.

closed 2019-11-14 by t-paul