openscad/openscad #2659: GCC warnings about changing behaviour.

rewolff, 2018-12-28:
GCC reports: 
```
In file included from /usr/include/c++/6/vector:69:0,
                 from src/GeometryUtils.h:4,
                 from src/polyset.h:6,
                 from src/polyset.cc:27:
/usr/include/c++/6/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_insert_aux(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {Eigen::Matrix<double
, 3, 1, 0, 3, 1>}; _Tp = Eigen::Matrix<double, 3, 1>; _Alloc = std::allocator<Eigen::Matrix<double, 3, 1> >]’:
/usr/include/c++/6/bits/vector.tcc:321:7: note: parameter passing for argument of type ‘std::vector<Eigen::Matrix<double, 3, 1> >::iterator {aka __gnu_cxx::__normal_iterator<Eigen::Matrix<double, 3, 1>*, std::vector<Eigen::Matrix<double, 3, 1> > >}’ will change in GCC 7.1
       vector<_Tp, _Alloc>::
       ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/6/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_insert_aux(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const Eigen::Matrix<double, 3, 1, 0, 3, 1>&}; _Tp = Eigen::Matrix<double, 3, 1>; _Alloc = std::allocator<Eigen::Matrix<double, 3, 1> >]’:
/usr/include/c++/6/bits/vector.tcc:321:7: note: parameter passing for argument of type ‘std::vector<Eigen::Matrix<double, 3, 1> >::iterator {aka __gnu_cxx::__normal_iterator<Eigen::Matrix<double, 3, 1>*, std::vector<Eigen::Matrix<double, 3, 1> > >}’ will change in GCC 7.1
/usr/include/c++/6/bits/vector.tcc: In member function ‘void PolySet::insert_vertex(const Vector3d&)’:
/usr/include/c++/6/bits/vector.tcc:128:8: note: parameter passing for argument of type ‘const __gnu_cxx::__normal_iterator<Eigen::Matrix<double, 3, 1>*, std::vector<Eigen::Matrix<double, 3, 1> > >’ will change in GCC 7.1
        _M_insert_aux(__pos, std::move(__x_copy));
        ^~~~~~~~~~~~~
/usr/include/c++/6/bits/vector.tcc:131:6: note: parameter passing for argument of type ‘const __gnu_cxx::__normal_iterator<Eigen::Matrix<double, 3, 1>*, std::vector<Eigen::Matrix<double, 3, 1> > >’ will change in GCC 7.1
      _M_insert_aux(__pos, __x);

```

I'm compiling on raspberry pi (raspbian). 
`gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
`
with manual adjustments (adding a few libs that the linker says should be later in the list) I got the whole thing to compile. It starts and shows the startup splash, and then segfaults. Recompiling with -g enabled. ...

t-paul, 2018-12-28:
Which code is that? Latest `master`? Release version?

rewolff, 2018-12-28:
Sorry. git pull from the repository here. I'm guessing master, yes. It seems I did something wrong with the network configuration of the computer/pi in question. Can't reach it now.

kintel, 2018-12-31:
This appears to be a common build warning for various projects on Raspberry Pi.
I'd be interested in knowing what, exactly, was changed in gcc-7.1.

rewolff, 2018-12-31:
It is behavior that WILL be changed, according to the message. 
The raspberry comes with: 
`gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
`

kintel, 2018-12-31:
..but what feature, and from what old to what new behavior? This must be a specific C++ feature, no?

rewolff, 2018-12-31:
Yeah. From the message I understand it is something to do with overloaded operators and stuff like that. C++ specific, I'm not good in the details on C++.