openscad/openscad #2660: CGAL assertion violation when adding primitives to STL

Pezmc, 2018-12-29:
I see forum threads and mailing list threads talking about this problem, but I'm still not entirely sure on how to fix it. I'm making a bunch of straightforward mods to an existing STL, the preview works fine, but on render it throws:

```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /Users/pezcuckow/Git/libraries/install/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330
```

The STL itself renders just fine on its own, it's only getting upset if there are other primitives in the scene.

As I understand from the FAQ this is likely a problem with the STL file, so as advised I ran it through Filters -> Cleaning and Repairing -> Remove T-Vertices by Edge-Flip and imported the new file, but I'm still seeing the same problem. 

Can anyone help me out here?

Files:

- [Seca_elefante_1_1_Repaired_with_Meshlab.stl](http://c.pez.io/9e987f8d4df7)
- [Seca_elefante_1_1_Original_Model.stl](http://c.pez.io/9f0aaa21178f)

### Minimal example that reproduces the error:

```openscad
import("Seca_elefante_1_1_Repaired_Mesh.stl", convexity = 5);
cube(5);
``` 

### Full code
```openscad
$fn = 10;

union() {
    import("Seca_elefante_1_1_Repaired_Mesh.stl", convexity = 5);

    // Flatten bottom
    translate([-45,-88,6])
        rotate([5,0,0])
        cube([10,40,7.5]);
        
    // Block trunj
    translate([-45,-89,6])
        rotate([5,0,0])
        cube([10,4,10]);

    topX = -78.5;
    topY = -88;

    height = 25;

    bottomX = 0;
    bottomY = 0;
               
    holeSize = 2.5;   

    offsetX = -1.5;
    offsetY = -0.5;
    
    difference() {
        // Bottom Plate
        translate([topX,topY,height])
            linear_extrude(2.5)
                square([78,80]);

        // Holes in bottom plate
        translate([0,0,height - 0.25])        
        linear_extrude(height + 0.5)
        for (x =[topX:holeSize*8:bottomX]) {
            for (y =[topY:holeSize*8:bottomY]) {
                translate([x + holeSize*4 + offsetX, y + holeSize*4 + offsetY])
                circle(holeSize, true);
            }  
        }
    }

    supportWidthOffset = 10;

    supportOffset = holeSize * 4;

    // Supports for base
    translate([0,0,7])
    linear_extrude(20)
    for (x =[topX:holeSize*16:bottomX - supportWidthOffset]) {
        for (y =[topY:holeSize*16:bottomY - supportWidthOffset]) {
            translate([x + holeSize*4 + offsetX + supportOffset, y + holeSize*4 + offsetY + supportOffset])
            circle(holeSize, true);
        }  
    }

    // Center support for base
    centerOffsetX = -1;
    centerOffsetY = 4.5;
    translate([(topX - bottomX) / 2 + centerOffsetX,(topY - bottomY) / 2 - centerOffsetY,7])
    linear_extrude(20)
    circle(holeSize, true);
}
```

nophead, 2018-12-29:
It looks fine in NetFabb too. I suspect it simply has vertices that are too close together for OpenSCAD. Try reducing the level of detail.

MichaelAtOz, 2018-12-29:
Both Meshlab & NetFabb show (different) Self Intersections:
[image]

[image]

I've always had trouble fixing those...

MichaelAtOz, 2018-12-29:
Note this issue hows when you do a CSG operation (union/difference etc) which invokes CGAL to process the STL geometry. 
When you see an error like:
CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle()
Basically it says CGAL is checking and it fails, in this case some face is != to another that should be the same.

Closing. Bad STL that CGAL can't process. 
Closing doesn't mean others can't comment further if they have suggestions.

nophead, 2018-12-29:
Oh, yes I didn't realise the free NetFabb doesn't show self intersections by default and can't fix them.