openscad/openscad #2663: Inconsistent CGAL error??

MichaelAtOz, 2018-12-29:
Looking at the Unicode stuff had me glancing all over the place, including looking at messages, errors which lead on to exit()'s.

Should this be exit(1)?

https://github.com/openscad/openscad/blob/75b873f29f9a9bdd382e7e6ebf338091de453b26/src/ext/CGAL/OGL_helper.h#L243
```
  inline void CGAL_GLU_TESS_CALLBACK errorCallback(GLenum errorCode)
  { const GLubyte *estring;
    estring = gluErrorString(errorCode);
    fprintf(stderr, "Tessellation Error: %s\n", estring);
    std::exit (0);
  }
```

Also, other CGAL errors are of the form
>std::cerr << "Error importing polygon" << std::endl;
exit(1);

Is the above fprintf correct?

kintel, 2018-12-30:
`OGL_helper.h` is a modified version of some CGAL code: https://github.com/CGAL/cgal/blob/master/Nef_3/include/CGAL/Nef_3/OGL_helper.h

This is indeed a bit suboptimal. CGAL itself doesn't appear to be using this any longer; they rewrite to using their own triangulator I think. Perhaps we should consider scrapping this as well...

MichaelAtOz, 2026-03-17:
Probably still needs consideration, "Perhaps we should consider scrapping this as well...".
Reopening.