openscad/openscad #2664: Accept 3D printing integration as a stable feature

kintel, 2018-12-30:
TODO:
* [ ] Handle Qt4 builds
* [ ] Resolve handling of async response (busy loop vs. dialog vs. background)
* [ ] Dialog for initial run (for enabling the feature)
* [ ] Managing upload size (too big, progress, cancel)

t-paul, 2019-01-07:
See #2688.

kintel, 2019-01-16:
In terms of shortcut, I like the suggestion to make `F7` STL export and `F8` 3D printing.
In the future we may implement toolbar editing to allow F7 to be bound to other export formats.

printathing, 2019-01-16:
> In terms of shortcut, I like the suggestion to make `F7` STL export and `F8` 3D printing.
> In the future we may implement toolbar editing to allow F7 to be bound to other export formats.

Agreed!

t-paul, 2019-01-16:
I'd leave the UI details till we got some feedback, everything else should be covered by #2708.

commit 6cdfb843 (2019-01-23): Remove experimental flag (fixes #2664).
closed 2019-01-23 by kintel