openscad/openscad #2673: Generated csg files have import paths relative to the source file

nophead, 2019-01-02:
When a csg file is exported in a different folder to the source file and is then opened in OpenSCAD it fails to find any imported files. This is because the file paths are all relative to the source file. 

Since it is generated by dumping the tree it is a correct representation of the internal data structure but to make a stand alone csg file the import paths need to be translated to be relative to where it is being generated.

kintel, 2019-01-02:
That's a good point. I assume that got added with the test framework in mind, but not sure how this would even work with the tests..

kintel, 2019-01-02:
..but when we fix this, we should probably make sure we have a test for it, so we don't re-break it again in the future

nophead, 2019-01-03:
Other than the test results needing re generating I don't think it would break them as long as they are still relative paths.