openscad/openscad #2674: Window menu for macOS user interface

rlaferla, 2019-01-03:
The macOS user interface desperately needs a Window menu to manage open windows.  It should contain "Minimize", "Zoom", "Bring All to Front" menu items.  If you open a .scad file that is already open, it needs to bring the already open file to the foreground and not open it a 2nd time.

[image]

Zenmaster43, 2021-03-15:
I concur with rlaferla on this. It's an incredible waste of time trying to locate the window I was just in before switching to another app and back, when I've got to multi-task on a major scale, with scores of windows open in each of many apps, spread over multiple spaces. I actually really like the way BBEdit does it, by using a single window for all documents and providing a document selection list in a bar on the left side of the window. This alleviates the need for a separate menu.

Sod-Almighty, 2024-11-26:
I agree. On my system, I can minimise the active window by swiping down with four fingers on my Apple trackpad; or – and I believe this is a standard system hotkey – by pressing CMD+M. This works for 99% of applications. Guess which application it _doesn't_ work with?

The reason it doesn't work is almost certainly because there is no "Minimize" option in the Window menu. Is it really so much to ask for OpenSCAD to respect the fundamental user interface standards of the platform on which it is running?

Sod-Almighty, 2026-03-18:
Why is this issue stale? Why is it stalled? It's not waiting on any information from the people who reported it. All it's waiting for is the developer to fix it.....after SEVEN BLOODY YEARS of being broken. Is it really too much to ask for the software to behave like any other MacOS application?

OpenSCAD is hard enough to use as it is, without putting extra usability hurdles between the software and the user.