openscad/openscad #2686: Snapping Polyset vertices to a grid can cause CGAL exceptions

nophead, 2019-01-07:
This code shows the problem with F5. Clearing the cache before F6 gives a correct result.
```
echo(version());
eps = 1/128;
layer_height = 0.25;

module hanging_hole(z, ir, h = 100, h2 = 100) {
    infill_angle =  45;
    render(convexity = 3) translate([0, 0, z]) {
        rotate(infill_angle)
                intersection() {
                    rotate(-infill_angle)
                        linear_extrude(height = 3 * layer_height, center = true)
                                //offset(-eps) 
                            children();

                    cube([100, 2 * ir, 2 * layer_height], center = true);
                }
        below = min(z + eps, h2);
        translate([0, 0, -below])
            linear_extrude(height = below)
                children();
    }
}

$fn = 3;
cube();
render()
    translate([20, 10])
        difference() {
            translate([0, 0, 5 / 2])
                cube([25, 25, 5], center = true);
            
            hanging_hole(z = 2, ir = 2.5)
                circle(r = 5 / cos(180 / $fn));
            
        }
```
```
ECHO: [2019, 1, 3]
Compiling design (CSG Products generation)...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426
```

It goes wrong because a rotated copy of a triangle stacked on top of itself has very small differences in the vertex coordinates. This leads to some very thin triangles when unioned. When these are snapped to a grid a thin triangle can become degenerate because all its points end up in a line and CGAL then rejects it. It can also get flipped as the winding order can change when the triangle is thin.

It works in F6 because the union is never converted to a Polyset. F5 caches a broken Polyset so F6 does not work after F5.

Polyset::quantize() needs to detect degenerate triangles but it can't simply remove them. It needs to stitch up the hole that is left. It also needs to fix the winding order in some cases as well. Possibly there are other topology breaking things quantization can do as well. It can certainly create self intersections by merging vertices that were previously distinct, so general mesh repair is needed.

MichaelAtOz, 2019-12-24:
Another example #3168

kintel, 2024-03-30:
Status as of 2024-03-24: F5 still broken when using CGAL. Works with Manifold.