openscad/openscad #2692: Linear_Extrude & Difference don't play nice

Digital-Corpus, 2019-01-09:
I'm attempting to create a louvered surface, both negative and positive relief, for a heat spreader idea I have. In order to minimize mismatch when a material is added between the two pieces, I'm opting to use a sine wave for the profile. To generate this type of geometry, I make a 2D poly and linearly extrude it applying it via difference to the plate I need. However, GCAL doesn't like this for some reason and being a n00b, I'm not sure why, despite a properly rendering preview.

My specific error:

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326

Here is my code:
```
$fn = 180;
//$vpr = [60,0,110];
//$vpt = [17,30,0];

module plate(xlen, ylen, zlen, rad_top, rad_bot) {
hull(){
        translate([rad_bot, rad_bot,0]){
            cylinder(r1=rad_bot, r2=rad_top, h=zlen);
        };
        translate([xlen-rad_bot, rad_bot,0]){
            cylinder(r1=rad_bot, r2=rad_top,  h=zlen);
        };
        translate([rad_bot, ylen-rad_bot,0]){
            cylinder(r1=rad_bot, r2=rad_top, h=zlen);
        };
        translate([xlen-rad_bot, ylen-rad_bot,0]){
            cylinder(r1=rad_bot, r2=rad_top, h=zlen);
        };
    };
};

module louvers(count, spread, amp, depth, fn, side){
    wave_profile = [ for(i = [0 : 360/fn : 360*count]) [i/360*spread/count, sin(i)*amp]];
//    echo(wave_profile);
    if (side == 1){
        difference(){
            linear_extrude(depth) polygon(points=wave_profile);
            translate([-0.0005,-0.0005,-0.0005]){
                cube([spread+0.001, amp+0.001, depth+0.001]);
            }
        }
    }
    else if (side == -1){
        difference(){
            linear_extrude(depth) polygon(points=wave_profile);
            translate([-0.0005,-amp+0.0005,-0.0005]){
                cube([spread+0.001, amp+0.001, depth+0.001]);
            }
        }
    }
}
module louvered_plate(xlen, ylen, zlen, rad_top, rad_bot, count, fn, side){
    spread = xlen;
    amp = zlen/2;
    depth = ylen;
    union(){
        difference(){
            color("silver")
            plate(xlen, ylen, zlen, rad_top, rad_bot);
            rotate([-90,0,0]){
                translate([0,+0.001,-0.0005]){
                        louvers(count,spread,amp,depth+0.001,$fn,side);
                }
            }
        }
        intersection(){
            rotate([-90,0,0]){
                louvers(count,spread,amp,depth-0.001,$fn,-side);
            }
            color("silver")
            translate([0,0,-amp]){
                plate(xlen, ylen, zlen, rad_top, rad_bot);
            }
        }
    }
}

translate([1.25,10.95,3.]){
    color("white")
    plate(31.5,46,1.018,3,3);
}

translate([1.25,10.95,]){
    color("silver")
    louvered_plate(31.5,46,1.018,3,3,8,$fn,1);
}
```

Digital-Corpus, 2019-01-09:
I can make a workaround using cylinders and hull, but I was expecting this to work.

nophead, 2019-01-09:
> Here is a simplified version that works:
>
> $fn = 180;
> //$vpr = [60,0,110];
> //$vpt = [17,30,0];
>
> module plate(xlen, ylen, zlen, rad_top, rad_bot)
> hull(){
>     for(side = [-1, 1], end = [-1, 1])
>         translate([end * (xlen / 2 -rad_bot), side * (ylen / 2 - rad_bot)])
>             cylinder(r1=rad_bot, r2=rad_top, h=zlen, center = true);
> }
>
> module louvers(count, spread, amp, depth, fn){
>     wave_profile = [[0, -1], for(i = [0 : 360/fn : 360*count])
> [i/360*spread/count, sin(i)*amp + amp],[spread, -1]];
> //    echo(wave_profile);
>     translate([-spread / 2, 0])
>         linear_extrude(depth, center = true, convexity = count)
> polygon(points=wave_profile);
> }
> module louvered_plate(xlen, ylen, zlen, rad_top, rad_bot, count, fn, side){
>     spread = xlen;
>     amp = zlen/2;
>     depth = ylen;
>     difference(){
>         plate(xlen, ylen, zlen * 2, rad_top, rad_bot);
>
>         translate([0, 0, - side * zlen])
>             rotate([side * 90,0,0])
>                 louvers(count,spread + 1/128,amp,depth + 1,$fn);
>     }
> }
>
> translate([1.25,10.95,3.]){
>     color("white")
>     louvered_plate(31.5,46,1.018,3,3,8,$fn,-1);
> }
>
> translate([1.25,10.95,]){
>     color("silver")
>     louvered_plate(31.5,46,1.018,3,3,8,$fn,1);
> }
>
> The original chopped up the sine wave into positive and negative halves
> and then put them together again. I think that created thin slivers doe to
> floating point inaccuracies, which OpenSCAD doesn't like. I added a base
> below the sine wave so it makes a single solid object that can be
> subtracted. I also centred everything as I find that easier to work with.

Swapping sin to cos gives better symmetry.

Digital-Corpus, 2019-01-09:
I'm not so concerned about symmetry since there is a slug that is offset from the center that I wish to align to the louvers. Thanks for the re-write of the script as it gives me a better understanding of the syntax of OpenSCAD, which is the learning curve I'm dealing with. I was hoping that aliasing $fn/fn to a divisor of 360 that I'd avoid the rounding issues and that extra clipping would remove such unpleasantness. Guess not.

Anyhow, in 2015.03, Win10, I'm getting a compilation/parsing error telling me that there is an issue with the syntax of the for loop in the maxtrix definition of wave_profile, line 13.

nophead, 2019-01-09:
I am surprised such an old version has list comprehensions at all.

Try this:
```
    wave_profile = concat(
        [[0, -1]], 
        for(i = [0 : 360/fn : 360*count]) [i/360*spread/count, cos(i)*amp + amp]],
        [[spread, -1]]
    );
```

Digital-Corpus, 2019-01-09:
No go. Same complaint about the open paren on the for loop.

nophead, 2019-01-09:
Sorry it got deleted when I formatted it. Should be
```
    wave_profile = concat(
        [[0, -1]], 
        [for(i = [0 : 360/fn : 360*count]) [i/360*spread/count, cos(i)*amp + amp]],
        [[spread, -1]]
    );
```

Digital-Corpus, 2019-01-09:
Good to go. The docs reference 2015.03 so I've been using it. Bumped over to the developmental versions as a result of this. Thanks again for the assist.

closed 2019-01-09 by Digital-Corpus