openscad/openscad #2699: Bail out of lengthy preview / render in a more graceful way

TLC123, 2019-01-13:
I keep finding myself in cycles of being unable to bail, force closing and reopening OpenScad, with scripts that unresponsive, taking way longer than intended, throwing hundreds of warnings but keep running or just plain freeze up the GUI.
Over and over usually via a trip to backupfolder to fetch  the latest version.

Would it be possible to bail out of lengthy preview / render in a somewhat more graceful way?
Or at least on startup after a crash get an option to automatically fetch the latest backup.

MichaelPFrey, 2019-01-13:
What operating system are you using?
Are you on a release or on a development build?

Recent development builds have a new option. If that option is enabled, openscad stops on the first warning.

t-paul, 2019-01-13:
Duplicate of #1195.

closed 2019-01-13 by t-paul

MichaelPFrey, 2019-01-13:
to me, it sounds more like #329

thehans, 2019-01-13:
For what its worth, during an F6 render, you should still be able to save the file with Ctrl-S,  its just the preview that locks the GUI at the moment.

TLC123, 2019-01-13:
Win7 OpenSCAD version 2019.01.10.ci1116
 Thanks for the tips
This might be interesting: Whenever i put an echo out inside a function to read out some debuginfo. The gui gets responsive for a tiny moment.  Don't know why that is. But usually enought to scroll down to the piece of code i'm interested in reading . Caveat is that deeply nested functions get a lot slower over all and it spams up the console. I wish I at least could look around for my mistake while the code compiles.

MichaelPFrey, 2019-01-13:
> This might be interesting: Whenever i put an echo out inside a function to read out some debuginfo. The gui gets responsive for a tiny moment. Don't know why that is.

I assume that echo is "somewhere down the line" calling `MainWindow::processEvents();`. We could develop an inbuilt function, which does nothing, except call `MainWindow::processEvents();` when in GUI mode (and does nothing in the console mode).

The other solution would be to evaluate the scad-code in a seperate thread. (Looking at the other related issues, that might take some time).

@kintel How would you think about the idea of giving users a nicely wrapped way to call `MainWindow::processEvents();` to give the user some control over the interactivity of the GUI during evaluation phase? I think of something like a builtin function called `nop` (as in **n**o **op**ereation), which takes one (optional) parameter, calls `MainWindow::processEvents();`and then returns the parameter (or undef).

An other name could be `pause();`, but pause returning its parameter does not make sense. A module would be less versatile. (a function can be used in a module, but a module can not be used in a function)

It should work, but it also is somewhat of an hack, that exposes things the user should not have to think about.

t-paul, 2019-01-13:
As the script evaluation should not depend on the GUI part, there's not much other choice than doing the evaluation in a separate thread. That is not the same as multi-threading though. So it's a bit of work (getting rid of the central GUI-Lock) but it's not in the same class as the multi-thread discussion.

t-paul, 2019-01-13:
Yes, it's an old issue, but that does not always mean it's complicated to do. There might be no agreement on how to solve it, or maybe just nobody had time to spend on the issue.

TLC123, 2019-01-13:
I might add that i have 6-8 OpenSCAD windows open, working in different files. The ones i open from within my fist widow gets unresponsive with the mother window but i can keep editing other ones that i opened separately . Isn't that a bit strange?