openscad/openscad #2700: Accept 3MF export/import as stable features

kintel, 2019-01-16:
This feature should be ready. Only task left is to remove it from experimental

nophead, 2019-01-16:
I don't think it works on Windows, all the tests fail.

t-paul, 2019-01-16:
Oh? I did test that not long ago and that was fine, hmm...

t-paul, 2019-01-16:
Right, as said before, there is no lib3mf package for MSYS2, so tests can't work there. The only criteria is the official MXE based build which shiould have it enabled.

kintel, 2019-01-16:
@nophead Is it possible to run the MSYS2 test suite using the MXE binary with some manual file copying you think? Asking you since you appear to own a Windows box :)

nophead, 2019-01-16:
I could have yesterday but now I am a few thousand miles nearer the equator for a month.

t-paul, 2019-01-16:
I'll have a look if my Win10 VM is still working.

t-paul, 2019-01-17:
Simple case works in interaction with Win10 3D Builder.

Export from OpenSCAD -> 3D Builder
[image]

3D Builder -> Import in OpenSCAD 
[image]

commit 95ed6ab8 (2019-01-23): Merge pull request #2716 from openscad/3mf-remove-experimental-flag

Remove experimental flag from 3MF import/export (fixes #2700).
commit cca8392f (2019-01-23): Remove experimental flag from 3MF import/export (fixes #2700).
closed 2019-01-23 by kintel