openscad/openscad #2702: Accept SVG import as a stable feature

kintel, 2019-01-16:
TODO:
* [ ] Decide on a minimal subset to support
* [ ] Add more tests
* [ ] Call out any isolated bugs as separate issues

MichaelPFrey, 2019-01-16:
I would like to ask, how you would like to handle errors/warnings within or related to libsvg.
(for e.g. file not found or parsing error)

Using printutils would go against the intention of it being standalone library. Leaves us with error codes or exceptions. Given that this is a c++ library and project, exceptions seam to be right mechanism.
Any thoughts? I have not yet looked deep into the code and I do not know the bigger picture, so I rather ask before starting to hack.

t-paul, 2019-01-17:
#2596 adds a number of tests, mostly based on the SVG spec example files and also activates import/export test cases. I think that should cover the core functionality (= importing SVG Path).

As for error reporting, I suppose exceptions are the way to go. However, I don't think we need too much with regard to detailed warnings/parse errors messages as I suspect SVG is rarely written by hand (even though the spec seems to assume that is the main use case for some reason).

Note that #2596 also clarifies this bunch of code to have a different license (which was intended from the start but never happened to end up stated in the files). It probably makes sense have the relatively invasive changes from this PR merged first (or putting smaller changes on top of this branch).

MichaelPFrey, 2019-01-17:
I do not want to add much error handling - but some like "file not found" and a generic "error during parsing" would make sense, as silently returning no geometry can cause hard to track down issues in bigger projects.

t-paul, 2019-01-17:
Yes, that sounds great.

t-paul, 2019-01-23:
closed via #2596

t-paul, 2019-01-23:
Oh, whops, the experimental flag was not part of the PR...

commit 8e55caf4 (2019-02-24): Remove experimental flag from SVG import (fixes #2702).
closed 2019-02-24 by kintel