openscad/openscad #2703: Accept Customizer as a stable feature

kintel, 2019-01-16:
TODO:
* [ ] Write TODO list

kintel, 2019-01-24:
@MichaelPFrey You may have a better idea what's absolutely needed for this to become stable, or is it already good enough?. We have two issue with a lot of suggestions: #1781 and #2316, that may need some cleanup

MichaelPFrey, 2019-01-26:
Good question. I currently see no blocking issue. The main benefit of the experimental status is, that the feature can quickly be disabled when it makes problem. I have most experimental features on all the time, so I assume that the customizer now plays pretty nicely with the other subsystems. I'll do some manual testing over the weekend

On thing I think I observed is, is that the customizer is creating unnecessary json files again. But I am not sure and even if: This is not a big deal.

kintel, 2019-01-26:
Thanks. I think investigating the JSON issue makes sense.  I've updated the TODO list.

t-paul, 2019-02-10:
I think the JSON file creation issue is fixed by https://github.com/openscad/openscad/pull/2776/commits/1aeb0a2b51532072a8a320e45be42ad4da7646d3.

commit 33c2e7e0 (2019-02-10): Remove experimental flag from customizer (fixes #2703).
closed 2019-02-19 by kintel