openscad/openscad #2704: Accept InputDriver as stable feature

t-paul, 2019-01-16:
TODO:

- [ ] Remove experimental flag for everything except the DBus driver

MichaelPFrey, 2019-01-16:
I like the simplicity of the todo list :smiley: The one for getting it merged was a lot longer - but off-course a development branch had its own benefits as it lead to a pretty clean interface to the other software parts, as the PR always showed the whole impact that inputdriver has/had on the whole.

DBus is kind of a developer tool right now anyway and I think it will remain that for a while.

It is awesome to develop drivers with it, as you don't have to recompile or even restart openscad.
It will be an awesome stepping stone for future drivers.
But given the fact that is aims it self at developers under linux, it remaining experimental is the right thing.

On a related side note: Any interest in adding https://github.com/MichaelPFrey/openscad-dbus-reference to the openscad project?

t-paul, 2019-01-17:
Sure, we can move the repo into the organization. That might make it a bit easier to discover. If I'm reading the docs right, we'd need to enable repo creation temporarily as that is normally disabled.

commit 2b505f1d (2019-01-22): Change input-driver experimental flag to cover only the DBus driver (fixes #2704).
commit d26ea931 (2019-01-22): Merge pull request #2710 from openscad/input-driver-remove-experimental-flag

Change input-driver experimental flag to cover only the DBus driver (fixes #2704).
closed 2019-01-22 by kintel