openscad/openscad #2705: Can't build master on Ubuntu 18.04

hexane360, 2019-01-16:
I've successfully built openscad in the past, but I'm having trouble with the current master.

Make gives the following error:
```
g++ -Wl,-O1 -o openscad    -lQt5Gui -lQt5Core -lGL -lpthread 
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/Scrt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
Makefile:66: recipe for target 'openscad' failed
make: *** [openscad] Error 1
```
scripts/check-dependencies.sh gives the following:
```
depname     minimum     found       OKness      
qt          4.4         5.9.5       OK          
qscintilla2 2.7         2.10.2      OK          
cgal        3.6         4.11        OK          
gmp         5.0         6.1.2       OK          
mpfr        3.0         4.0.1       OK          
boost       1.35        1.65.1      OK          
opencsg     1.3.2       1.4.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.3.4       OK          
glib2       2.0         2.56.3      OK          
fontconfig  2.10        2.12.       OK          
freetype2   2.4         2.8.1       OK          
harfbuzz    0.9.19      1.7.2       OK          
libzip      0.10.1      1.1.2       OK          
bison       2.4         3.0.4       OK          
flex        2.5.35      2.6.4       OK          
make        3           4.1         OK
```

Any help would be appreciated

t-paul, 2019-01-16:
That is not enough information to help. Please link the full build log. The error seems strange and more related to a broken build environment than OpenSCAD.

hexane360, 2019-01-16:
That *was* the full build log.

I found the problem. I was trying to use `qmake openscad.pro.user` to build with saved custom options. However, that wasn't including `openscad.pro` as well.

closed 2019-01-16 by hexane360