openscad/openscad #2707: Implement protocol to allow OpenSCAD to run in Jupyter notebooks

WillAdams, 2019-01-16:
There was a bit of discussion on the mailing list/forum:

http://forum.openscad.org/Jupyter-Notebook-integration-was-Hosting-OpenSCAD-Files-was-Re-Is-it-possible-to-write-out-a-text-fi-td24108.html

See: https://github.com/jupyter/jupyter/wiki/Jupyter-kernels

Being able to do 3D modeling in Jupyter notebooks would be amazing, and would help greatly in using Jupyter as a notebook for machining and CNC and 3D printing.

t-paul, 2019-01-16:
I actually did look at that topic some time ago because I think it's interesting. Still it was not really clear what specifically is needed. So if someone can provide a more detailed description of what would be needed in OpenSCAD, that would hugely increase the chance of anything getting implemented.

WillAdams, 2019-01-17:
Does it help that CADquery has an extension for this? https://github.com/bernhard-42/cadquery-jupyter-extension

pschatzmann, 2020-02-27:
I wanted to do do my OpenSCAD Designs in Jupyterlab as well, so I started to work on a solution. Details can be found at https://www.pschatzmann.ch/home/2020/02/26/an-openscad-kernel-in-jupyter/

Any feed-back is welcome.

t-paul, 2020-03-12:
@pschatzmann This is an interesting solution. Being able to actually insert OpenSCAD code into cells is certainly something I would see as mandatory for calling it an OpenSCAD kernel.

How is that parser handling arbitrary OpenSCAD code, specifically new syntax? There's quite a wishlist of things that could be added to the language to make it more powerful, like the new function literals (included in nightly build), array slicing (exists only as prototype) and much more.

pschatzmann, 2020-03-12:
Very high level: The parser tries to build a collection of all the functions, modules and top level assignments and replaces them with the updated definitions when a cells is calculated. 
In order to generate the STL or PGN the OpenSCAD code  is extracted into a file and I call the openscad command line functionality.

New syntax is unlikely to affect the functionality since I don’t care what’s within a module or function. 
I am still trying to find the corner cases where this concept might fall apart, but so far it looks quite promising...