openscad/openscad #2712: Broken translations?

baerbock, 2019-01-22:
I've tried to compile the current git version. It fails when it comes to the translations:
```
Compiling language files (CWD = ./openscad-git/src/openscad)...
msgfmt -c -v -o ./locale/fr/LC_MESSAGES/openscad.mo ./locale/fr.po
263 übersetzte Meldungen, 22 ungenaue Übersetzungen, 97 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/ru/LC_MESSAGES/openscad.mo ./locale/ru.po
300 übersetzte Meldungen, 11 ungenaue Übersetzungen, 71 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/de/LC_MESSAGES/openscad.mo ./locale/de.po
368 übersetzte Meldungen, 1 ungenaue Übersetzung, 13 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/cs/LC_MESSAGES/openscad.mo ./locale/cs.po
260 übersetzte Meldungen, 24 ungenaue Übersetzungen, 98 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/es/LC_MESSAGES/openscad.mo ./locale/es.po
233 übersetzte Meldungen, 66 ungenaue Übersetzungen, 83 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/pl/LC_MESSAGES/openscad.mo ./locale/pl.po
281 übersetzte Meldungen, 17 ungenaue Übersetzungen, 84 unübersetzte Meldungen.
msgfmt -c -v -o ./locale/uk/LC_MESSAGES/openscad.mo ./locale/uk.po
314 übersetzte Meldungen, 11 ungenaue Übersetzungen, 57 unübersetzte Meldungen.
Error: 4 unsupported relocations
Error: 4 unsupported relocations
convert: MagickCore/opencl.c:1495: CopyMagickCLCacheInfo: Zusicherung »pixels == info->pixels« nicht erfüllt.
```

MichaelPFrey, 2019-01-22:
The translations look fine by me. To me, the last three lines look more relevant:

```
Error: 4 unsupported relocations
Error: 4 unsupported relocations
convert: MagickCore/opencl.c:1495: CopyMagickCLCacheInfo: Zusicherung »pixels == info->pixels« nicht erfüllt.
```

and the assertion failed is a runtime error:

https://github.com/ImageMagick/ImageMagick/blob/fd7794d7de6a481fc3086fe496c95476fd473ee2/MagickCore/opencl.c#L1495

What operating system are you using? Are using cmake or qmake? Have you set any special options?
Could it be that you have build artefacts from an older build? This can for e.g. happen when you upgrade you distribution.

t-paul, 2019-03-11:
Closing due to inactivity. Feel free to reopen to continue the discussion...

closed 2019-03-11 by t-paul

baerbock, 2019-03-12:
@t-paul You are right, thanks.