openscad/openscad #2719: Display problems with Qt5 3D rendering

gskillas, 2019-01-23:
Hello together

openscad is really good and I appreciate the work you are doing in making it continously better. I have an issue here which bugs me as I work a lot remote using VNC to access a potent workstation (CGAL is quite taxing on computational resources, but very robust). When I compile with Qt5 under openSUSE Leap 15.0 everything compiles, but in the VNC session I am not able to interactively see what is being rendered. The rendering "frame" is just black. No coordinate system or anything showing. under display :0 everything works.

When using Qt4 (as far as this is possible) the 3D rendering works also over VNC (it uses MESA to do software rendering, but it works). What can I do about it?

Best regards & thank you for your time

George

t-paul, 2019-01-23:
That does not sound like an OpenSCAD issue. I suspect it means that this VNC stack is not able to forward accelerated OpenGL contexts. I don't see anything which can be done on OpenSCAD side.

Maybe check if one of the [Mesa Environment Variables](https://www.mesa3d.org/envvars.html) helps, e.g. LIBGL_ALWAYS_SOFTWARE.

gskillas, 2019-01-23:
Dear Paul,

mesa is runing fine. Paraview is working withpout problems. I suspect the issue is with Qt.

best,

George

kintel, 2019-01-23:
The ParaView GUI is written in Qt5 as well no? ..so it should be possible to get a side-by-side comparison by building both ParaView and OpenSCAD against the same version of Qt.
-> a pretty time consuming task though.

@gskillas What OS platforms, VNC server, VNC client, and GPUs are you using?

t-paul, 2019-01-23:
Grepping the paraview sources, I can't see any `QOpenGLWidget` or `QGLWidget`, so it's probably no reference.

gskillas, 2019-01-24:
> Grepping the paraview sources, I can't see any `QOpenGLWidget` or `QGLWidget`, so it's probably no reference.

You need to look at VTK:
ParaView-5.5.0 >grep -R QOpenGLWidget *
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:            // We do not call QOpenGLWidget::makeCurrent() as that also makes the
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:            // frame buffer object used by QOpenGLWidget active. This can have
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:  this->setUpdateBehavior(QOpenGLWidget::PartialUpdate);
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:  // QOpenGLWidget::resized() is triggered when the default FBO in QOpenGLWidget is recreated.
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:    // QOpenGLWidget::format() after the context is created. That's because
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:    // QOpenGLWidget always created context to samples=0.
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:    // When a QOpenGLWidget is told to use a QSurfaceFormat with samples > 0,
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:    // QOpenGLWidget doesn't actually create a context with multi-samples and
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:  // blit from this->FBO to QOpenGLWidget's FBO.
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:  // QOpenGLWidget says when this slot is called, the context may not be current
VTK/GUISupport/Qt/QVTKOpenGLWidget.cxx:      // widget. In that case, since QOpenGLWidget rendering into an offscreen
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * @brief QOpenGLWidget subclass to house a vtkGenericOpenGLRenderWindow in a Qt
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * QVTKOpenGLWidget extends QOpenGLWidget to make it work with a
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * manage the OpenGL context using QOpenGLWidget (added in Qt 5.4).
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * This is necessary since QOpenGLWidget wants to take over the window management as
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * Since QVTKOpenGLWidget uses QOpenGLWidget to create the OpenGL context,
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * it uses QSurfaceFormat (set using `QOpenGLWidget::setFormat` or
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * In QOpenGLWidget (superclass for QVTKOpenGLWidget), all rendering happens in a
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * QWidget subclasses (including `QOpenGLWidget` and `QVTKOpenGLWidget`) display
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * `QOpenGLWidget` subclasses are expected to do OpenGL rendering in
VTK/GUISupport/Qt/QVTKOpenGLWidget.h: * `QOpenGLWidget::paintGL`. QWidget can receive paint events for various
VTK/GUISupport/Qt/QVTKOpenGLWidget.h:#include <QOpenGLWidget>
VTK/GUISupport/Qt/QVTKOpenGLWidget.h:class VTKGUISUPPORTQT_EXPORT QVTKOpenGLWidget : public QOpenGLWidget
VTK/GUISupport/Qt/QVTKOpenGLWidget.h:  typedef QOpenGLWidget Superclass;
VTK/GUISupport/Qt/QVTKOpenGLWidget.h:   * Since by default `QOpenGLWidget::UpdateBehavior` is set to
VTK/GUISupport/Qt/QVTKOpenGLWidget.h:   * QOpenGLWidget::PartialUpdate, this means whatever was rendered in the frame

gskillas, 2019-01-24:
> The ParaView GUI is written in Qt5 as well no? ..so it should be possible to get a side-by-side comparison by building both ParaView and OpenSCAD against the same version of Qt.
> -> a pretty time consuming task though.
> 
> @gskillas What OS platforms, VNC server, VNC client, and GPUs are you using?

OpenSUSE LEAP 15.0 RealVNC Server, NVIDIA with nouveau (on display :0) and RealVNC on other displays. Everything works under display :0 . Under VNC the following image emerges:

I have compiled paraview 5.4 with Qt4 (fails Qt5), and paraview 5.6 with Qt5 (fails Qt4). Version 5.4 works, version 5.6 fails reporting GL and GLSL incompatibilities.
Interestingly 5.5 fails with both Qt4 and Qt5. All compiled and tested on the same system as openscad, which now compiles under Qt4 and Qt5 and works (GUI) under Qt4. 

Also I have set, as suggested

> export LIBGL_ALWAYS_SOFTWARE="true"

for MESA, it does not seem to have any influence. So to sum it up under VNC only Qt4 seems to work, Qt5 fails under VNC. I will confront the VNC guys with this as well, to get their perspective.

Best regards,

George

kintel, 2019-01-24:
Thanks @gskillas - I feels like this could be boiled down into testing against a VTK Qt demo app. That would also make it possible to report this against VTK (who have significantly more resources than us to hunt down bugs; and perhaps we/you can learn something that helps us too).

closed 2026-03-20 by github-actions[bot]