openscad/openscad #2729: Make the render sound only play during intensive operations

btlvr, 2019-01-27:
I just updated OpenSCAD, and I see that it plays a tune when the CGAL preview has finished rendering. 

**Pros of playing the sound:**
* It gets my attention if I walk away from my computer during a long render, or switch windows while waiting

**Cons of playing the sound:**
* It's mostly annoying and unnecessary. I don't need to be audially reminded every time I press F6.

My proposed solution is to only play the sound if the render has taken a certain amount of time (this would be an option in the preferences)

This would eliminate useless chimes, but still get the users attention after a long render.

MichaelAtOz, 2019-01-27:
There is a preference to turn it off BTW.

MichaelAtOz, 2019-02-20:
Perhaps the default should be False?

https://github.com/openscad/openscad/blob/ac1d89f2381f8d4b6eaf0492c8ba1666d7d7f393/src/Preferences.cc#L165

RomitKumar, 2019-03-13:
Implemented the feature at PR #2864

closed 2019-04-07 by kintel